Xinyuan-LilyGO/T-Display-S3 #82: What is exact difference between standard TFT_eSPI library and the one supplied with T-Display-S3

aksamdar, 2023-01-06:
It is hard to replace the TFT_eSPI library again and again.  The one supplied with T-Display-S3 doesn't work with T-Embed.  Standard TFT_eSPI works fine with T-Embed.  I am using both products and its hard to replace the TFT_eSPI library each time i am using a different board.  How can we solve this?

svdrummer, 2023-01-06:
This is messy for all of us. Maybe they should have two libraries with different names

mmMicky, 2023-01-07:
I have submitted a pull request to TFT_eSPI. After waiting for the merge, it can be used along with the main line. You can directly use the TFT library in the T-DISPLAY-S3 repository before it is merged.

svdrummer, 2023-01-14:
I understand the main library has been updated. I didn't see any instructions in the new library use. I will check the lib and assume there is one one more #defunes etc

ElectricDream5, 2023-01-17:
Nice update and improvements by Bodmer with TFT_ESPI 2.50. Among other things, a new user select profile for LilyGo T Display S3 (170x320) was created with id 206. This works well on Arduino IDE 2.03 on Windows 10 for me. I also selected TFT Inversion ON.

closed 2023-01-30 by mmMicky