Xinyuan-LilyGO/T-Display-S3 #84: How to convert images to C array ?

tavdog, 2023-01-13:
I've been trying everything and I can't get proper colors.  The color is always shifted one way or the other. see photo.
Is there a walkthrough somehwhere on how to do this ?  The tool in the TFT_eSPI doesn't work.  I can get recognisable images by exporting from GIMP as PNG and using img2cpp utility to create the C-array, but the colors are never correct.
[image]

tavdog, 2023-01-13:
I have tried the instruction in this repo but no matter what Blue and Red are swapped.
[image]

teastainGit, 2023-01-13:
It looks exactly like a 
tft.setSwapBytes(true);
issue, where the two colour bytes are swapped, Big Endian-Little Endian issue, where Ox00FF is transposed to 0xFF00

mmMicky, 2023-01-14:
Method 1: tft.setSwapBytes(true);
Method 2: Use the mouse to drag the blue and red to exchange

[image]

closed 2023-01-14 by mmMicky

tavdog, 2023-01-14:
Wow, thank you for the solution.  Dragging the Red block over to the blue fixed the issue.  But It never worked when I tried modifying swapBytes(true);  Why would setting swapBytes(false) not work, but dragging the little color swatch would ?

tavdog, 2023-01-14:
The other solution that works is changing the following options in the user setup for tft_espi
// #define TFT_RGB_ORDER TFT_RGB  // Colour order Red-Green-Blue
#define TFT_RGB_ORDER TFT_BGR // Colour order Blue-Green-Red

mmMicky, 2023-01-14:
swapBytes() is to modify the order of sending.
For example, the normal transmission is from bit7->bit6->....->bit0.
After enabling, the order is reversed even if bit0->bit1->..bit7