Xinyuan-LilyGO/T-Display-S3 #85: Question: PINs and SDA and SCL

hammikb, 2023-01-14:
I did not specifically see in in the pin lay out for any pins that will do SDA or SCL. Also know any digital pin can be used for those feature but just not really understand the pins. So please if anyone know or knows if it can work with this sensor HiLetgo GY-906 MLX90614ESF please let me know what I need to do. Thank you!

teastainGit, 2023-01-14:
I created a support and setup GitHub Repository for the LilyGO T-Display S3, here: 
(https://github.com/teastainGit/LilyGO-T-display-S3-setup-and-examples) 
The quick answer is that the JST connector beside the USB is the I2C port.
From the USB it is: G, 3.3V, SDA 43, SCL 44

[image]

-Terry

hammikb, 2023-01-16:
@teastainGit That you so much, also would you know how to call to the BOT button? Specifically im trying to use the bot button to turn the device on and off since i have it on battery

hammikb, 2023-01-16:
@teastainGit thank you so much! Do you by change know of a way to turn the device on and off with the Bot button once its on a battery source?

hammikb, 2023-01-18:
Any reason this shouldn't work? i have even tried Wire.begin(43, 44);

```

#include <Wire.h>
#include <Adafruit_MLX90614.h>

Adafruit_MLX90614 mlx = Adafruit_MLX90614();

void setup() {
 Serial.begin(9600);
 Serial.println("Adafruit MLX90614 test");  
 mlx.begin();   
}

void loop() {
 Serial.print("Ambient = "); Serial.print(mlx.readAmbientTempC());
 Serial.print("*C\tObject = "); Serial.print(mlx.readObjectTempC()); Serial.println("*C");
 Serial.print("Ambient = "); Serial.print(mlx.readAmbientTempF()); 
 Serial.print("*F\tObject = "); Serial.print(mlx.readObjectTempF()); Serial.println("*F");
 Serial.println();
 delay(500);
}
```

hammikb, 2023-01-18:
nvm i got it to work out of no where idk what i did

teastainGit, 2023-01-18:
@hammikb You do need 
Wire.begin(43, 44);  //SDA, SCL
in the setup loop to tell the ESP which pins are which.
Try this again and tell me what results do  you get with the sensor wired up and using this Wire.begin(43, 44); in the setup.
You should check out my repository and look at the  
HotHead_temp_sensor.ino 
which has an air temp sensor using I2C, but no library, just direct code so that you can see how it works.
(My code will only work with my sensor, this is just an example!)

hammikb, 2023-01-18:
Yeah thank you so much you example has actually helped me a lot! Let me know if you do know how to turn off these power pins to the IC2 thank you again!!!

uzyx-john, 2023-02-04:
Can pins 43 and 44 on the breadboard be used? I think they are connected.

teastainGit, 2023-02-04:
> [uzyx-john](https://github.com/uzyx-john) and [hammikb](https://github.com/hammikb)
Pins 43 and 44 are brought out at the 4 pin JST connector beside the USB aaaand on the side as per the pinout image.
Very good observation, keen eye, john.

lewisxhe, 2023-03-13:
If you have any questions, please open it again

closed 2023-03-13 by lewisxhe