Xinyuan-LilyGO/T-Display-S3 #86: Working with platformio, RTC_SW_CPU_RST reboot loop when using -DARDUINO_USB_CDC_ON_BOOT=1.

tavdog, 2023-01-16:
I don't know why but my board goes into reboot loop with the when CDC_ON_BOOT=1 is set.  It does not reboot loop when set to 0.  Here is the output log : 

Tron-68B6B3231E14
in tdisplay_s3 Constructor
out of Constructor
in tdisplay setup
[  1426][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
E (1193) gpio: gpio_set_level(226): GPIO output gpio_num error
[  1427][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
E (1199) gpio: gpio_set_level(226): GPIO output gpio_num error
going for tft.println(F("Test"))
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x420b9b6a
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x43c
load:0x403c9700,len:0xbec
load:0x403cc700,len:0x2a3c
SHA-256 comparison failed:
Calculated: dcde8d8a4817d9bf5d5d69a7247667264e4e10ac7493514868b61f5aa6146539
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403c98d8
Tron-68B6B3231E14

Here is platformio.ini

;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[platformio]
globallib_dir = ../../lib
src_dir = .

[env:ESP32-S3-DevKitC-1]

platform = espressif32
board = esp32-s3-devkitc-1
framework = arduino

platform_packages =
    framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#2.0.5

build_flags = 
    -DLV_LVGL_H_INCLUDE_SIMPLE
    -DBOARD_HAS_PSRAM
    -DARDUINO_USB_MODE=1 
    -DARDUINO_USB_CDC_ON_BOOT=1 

board_build.arduino.memory_type = qio_opi

mmMicky, 2023-01-30:
The first thing that needs to be ruled out is the code problem. You need to compile an empty program and keep printing in the loop to judge whether it is normal.

lewisxhe, 2023-03-13:
If you have any questions, please open it again

closed 2023-03-13 by lewisxhe