Xinyuan-LilyGO/T-Display-S3 #88: Deep Sleep

hammikb, 2023-01-19:
Hello

i am currently trying to find the best way to turn "off" my device when connected to a battery. The conclusions i have came to is the best way is to put the device into a deep sleep mode which will take down power usage dramatically. I have figured out how to turn off the screen so far but as far as putting the device into a deep sleep and waking it up im unsure. I have read some of the documents and was wondering if the T-display-s3 has any pins that work in that mode? Also is it possible to use the buttons already on the board to put it asleep and wake it up?

tyeth, 2023-01-19:
Look at the factory arduino sketch in the examples folder (you want to set your wifi settings in pin_config.h before building and uploading to get passed the wifi screen to the demo). It sets up one pin (button1) as a wake alarm (pin triggered interrupt) and the code just defines the wake reason then goes to sleep.  https://github.com/Xinyuan-LilyGO/T-Display-S3/blob/main/example/factory/factory.ino#L234-L243

You can do more complex things when only doing a light asleep, but start there (a deep sleep waiting for a pin to wakeup). I would advise looking at waking after a set time (number of seconds) have elapsed, and then once you have that working add in the pin wakeup with the time wakeup. You will then have covered RTC (realtime clock) and external0 input combined. There is an external1 input which supports more than one pin and is obviously going to help you in the future. In the docs there is documentation for the esp-idf framework (can be used by other frameworks like arduino / python) which has [more detail](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/sleep_modes.html), and then more usefully for you the esp-arduino docs https://espressif-docs.readthedocs-hosted.com/projects/arduino-esp32/en/latest/api/deepsleep.html
The other useful info is there is RTC memory which survives deep sleep, 8kilobytes worth, which means if you stick the right word prefix before variables, then they will be saved and available upon wakeup. Have a look at other peoples code for inspiration.

Long term, there is support for measuring ADC using the ULP core (lightweight CPU available in light-sleep), which means you could save even more by only waking to a light sleep to measure the battery voltage and then return to sleep if all is well.

hammikb, 2023-01-20:
@tyeth Thank you ill look at the factory code. I do need it to turn on and off though, not wake up after a certain time. Ill look more into the details on espressif

teastainGit, 2023-01-20:
Tyeth, thanks so much for the tip to stop the stupid ESPtouch function. The iPhone app would not connect and so I could not get the cool flip-clock.Works fine with manual entry of ssid and password.Thanks again.Cheers !Terry

closed 2023-03-13 by lewisxhe