Xinyuan-LilyGO/T-Display-S3 #89: BIN file to flash LilyGO T-Display-S3 to factory settings

ravco99, 2023-01-22:
Would it be possible to post to the repo the original factory bin file for this device?   Using the Flash Download Tool 3.9.3 from Espressif it should be possible to reload this device provided the correct BIN is available.  Other boards seem to have their BINs posted to the repo. 

Any other ideas to factory reset/re-flash ??   Is it possible to read the flash from a brand new board to create the bin ??

mmMicky, 2023-01-30:
I uploaded the firmware used for testing.

https://github.com/Xinyuan-LilyGO/T-Display-S3/tree/main/example/factory/firmware

closed 2023-01-30 by mmMicky

ravco99, 2023-01-30:
Thank you so much. Can you confirm which flash tool you used and which bin file ??

ravco99, 2023-02-13:
Would you be able to explain with screen shots how to use the 3.9.3 Flash
Download tool to use the bin files to restore this to factory setting?

there are 4 bin files
bootloader
partitions
firmware
boot_app0

I have followed the bat file and tried the following.   The chip does not
restart after this flashing.....

[image: image.png]

mmMicky, 2023-02-15:
https://github.com/Xinyuan-LilyGO/T-Display-S3/blob/main/example/factory/firmware/click_me.bat
There are different .BIN files uploaded to flash addresses here. The FLASH size information cannot be selected when using the ESPTOOL.PY tool. cause a reset. It is recommended to use the script method to write.

ravco99, 2023-02-16:
Micky......Thank you very much, I am getting closer but things are still
failing:
When i use ESPTOOL.PY on a Windows machine from a CMD window I get the
following

esptool.py --chip esp32s3   --baud 921600 --before default_reset --after
hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size
16MB 0x0 bootloader.bin 0x8000 partitions.bin 0xe000 boot_app0.bin 0x10000
firmware.bin
usage: esptool write_flash [-h] [--erase-all] [--flash_freq
{keep,80m,60m,48m,40m,30m,26m,24m,20m,16m,15m,12m}]
                           [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size
{detect,keep,256KB,512KB,1MB,2MB,2MB-c1,4MB,4MB-c1,8MB,16MB,32MB,64MB,128MB}]
                           [--spi-connection SPI_CONNECTION]
[--no-progress] [--verify] [--encrypt]
                           [--encrypt-files <address> <filename> [<address>
<filename> ...]]
                           [--ignore-flash-encryption-efuse-setting]
[--force] [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>:
*Detected overlap at address: 0x8000 for file: partitions.bin*

So it seems there is an error with the size of the bin files ???
OVerlapping addresses???

I can get esptool to connect to the chip and it shows 40Mhz so I changed
the command to 40m

Thoughts??

mmMicky, 2023-02-16:
Maybe you can use my .BAT script file directly. You just need to change the name.