Xinyuan-LilyGO/T-Display-S3 #90: Win10 upload to T-Display-S3 fails

gert4git, 2023-01-25:
Hi,

Using Win10 Arduino IDE 1.8.19 to program T-Display-S3. It worked and I was able to program the device until some time back. (My memory is not perfect what actually happened, but I think it might have been a windows update, or I just made a mistake) Anyway. Since now the code upload fails with message.

`Arduino: 1.8.19 (Windows 10), Board: "ESP32S3 Dev Module, Integrated USB JTAG, OPI PSRAM, QIO 80MHz, 16MB (128Mb), Core 1, Core 1, Hardware CDC and JTAG, Enabled, Disabled, Disabled, UART0 / Hardware CDC, Huge APP (3MB No OTA/1MB SPIFFS), 240MHz (WiFi), 115200, None, Disabled"
Sketch uses 714273 bytes (22%) of program storage space. Maximum is 3145728 bytes.
Global variables use 44972 bytes (13%) of dynamic memory, leaving 282708 bytes for local variables. Maximum is 327680 bytes.
C:\Users\gottsch\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\4.2.1/esptool.exe --chip esp32s3 --port COM7 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size 16MB 0x0 C:\Users\gottsch\AppData\Local\Temp\arduino_build_151596/TDispS3_Clock_gg.ino.bootloader.bin 0x8000 C:\Users\gottsch\AppData\Local\Temp\arduino_build_151596/TDispS3_Clock_gg.ino.partitions.bin 0xe000 C:\Users\gottsch\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6/tools/partitions/boot_app0.bin 0x10000 C:\Users\gottsch\AppData\Local\Temp\arduino_build_151596/TDispS3_Clock_gg.ino.bin 
esptool.py v4.2.1
Serial port COM7
A fatal error occurred: Could not open COM7, the port doesn't exist
A fatal error occurred: Could not open COM7, the port doesn't exist
`

There is a chance that I mis-remember my own steps which worked fine uploading code some time back. 

I am following instruction to push Boot button for upload code. 
I have installed SiLabs driver CP210x_Universal_Windows_Driver.zip (But that was not needed before) and today it does not help fix the issue.
It says it can't open the COM7 port. But the port is shown in the windows10 device manager.  Also the 'Get Board Info' shows info from the ESP32 that seems OK.

What am I missing?

Thanks,
Gert

teastainGit, 2023-01-25:
Hi Gert, my typical "long answer" is to goto my LilyGO T-Display S3 set up and examples repository, read the whole damned thing and follow step by step because this board is very persnickety and the first step is to get it running my simple HotHead demo and go from there. 
https://github.com/teastainGit/LilyGO-T-display-S3-setup-and-examples

gert4git, 2023-01-27:
@testainGit,

Thanks for the info. In the link I see 3 work folders with .ino files. Is there also a maintained document with detailed bring-up instructions of the board. Like a PDF with illustrations?

The issue was resolved as follows:

I had worked on an ATTiny project with tool chain in a kubuntu 22 VM running in 'WM Ware Player' on the Win10 laptop. The COM7 port had been given to the Atmel programmer and passed through to the VM. Programming worked well. Turns out that 'VM Ware Player' keeps a background service running even when the actual VM is not active and that service locks the COM port. Thus esptool.py failed to grab it, even though Arduino IDE was able to show board information on the same port.

As a fix I went into the Win 10 device manager and re-assigned COM10 to the ESP device. Then the esptool.py worked just fine.

You may want to add such tips and tricks to your documentation to help others in similar failure situations.

Thanks,
Gert

teastainGit, 2023-01-27:
Hi Gert, I'm trying to not sound defensive here(!) but my repository does include a lengthy text file called:
LilyGO T-Display-S3 Setup.txt
with tips and snags! I also include some reference to other users who contribute.
Your problem seems unique, but I am glad you got it running!
Cheers, Terry

yeguorong418, 2023-02-23:
Power off the mainboard, hold down the Key key, power on the mainboard for more than 1 second, and then release the key

lewisxhe, 2023-03-13:
If you still have problems, please open it again

closed 2023-03-13 by lewisxhe

gert4git, 2023-03-14:
All is great. It's a nice product.Cheers,Gert