Xinyuan-LilyGO/T-Display-S3 #91: Error compiling for board ESP32S3 Dev Module.

leetut, 2023-01-28:
[11670] Error loading Python lib '/var/folders/kz/yg8fjhp53c3201sw0bhpccy40000gn/T/_MEIkvqeyi/libpython3.8.dylib': dlopen: dlopen(/var/folders/kz/yg8fjhp53c3201sw0bhpccy40000gn/T/_MEIkvqeyi/libpython3.8.dylib, 10): Symbol not found: ____chkstk_darwin
  Referenced from: /var/folders/kz/yg8fjhp53c3201sw0bhpccy40000gn/T/_MEIkvqeyi/libintl.8.dylib (which was built for Mac OS X 11.0)
  Expected in: /usr/lib/libSystem.B.dylib
 in /var/folders/kz/yg8fjhp53c3201sw0bhpccy40000gn/T/_MEIkvqeyi/libintl.8.dylib
exit status 255
/Applications/Arduino.app/Contents/Java/arduino-builder returned 255
Error compiling for board ESP32S3 Dev Module.

mmMicky, 2023-01-30:
I can't tell from the message why the compilation failed. But first you need to determine whether you have installed the ESP32 SDK support package.

leetut, 2023-01-30:
I'm on osx 10 not 11
Could it be that?
Installed esp32 from boards manager, copied tft_esp32 folder into correct location

mmMicky, 2023-01-30:
This operation is correct. Please compile other examples in the example in the esp32 SDK package to confirm that the esp32 is installed completely.

leetut, 2023-01-30:
/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/lee/Library/Arduino15/packages -hardware /Users/lee/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/lee/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/lee/Documents/Arduino/libraries -fqbn=esp32:esp32:esp32s3:JTAGAdapter=default,PSRAM=disabled,FlashMode=qio,FlashSize=4M,LoopCore=1,EventsCore=1,USBMode=hwcdc,CDCOnBoot=default,MSCOnBoot=default,DFUOnBoot=default,UploadMode=default,PartitionScheme=default,CPUFreq=240,UploadSpeed=921600,DebugLevel=none,EraseFlash=none -ide-version=10813 -build-path /var/folders/kz/yg8fjhp53c3201sw0bhpccy40000gn/T/arduino_build_129162 -warnings=none -build-cache /var/folders/kz/yg8fjhp53c3201sw0bhpccy40000gn/T/arduino_cache_599257 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=/Users/lee/Library/Arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=/Users/lee/Library/Arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=/Users/lee/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch5 -prefs=runtime.tools.xtensa-esp32-elf-gcc-gcc8_4_0-esp-2021r2-patch5.path=/Users/lee/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch5 -prefs=runtime.tools.riscv32-esp-elf-gcc.path=/Users/lee/Library/Arduino15/packages/esp32/tools/riscv32-esp-elf-gcc/gcc8_4_0-esp-2021r2-patch5 -prefs=runtime.tools.riscv32-esp-elf-gcc-gcc8_4_0-esp-2021r2-patch5.path=/Users/lee/Library/Arduino15/packages/esp32/tools/riscv32-esp-elf-gcc/gcc8_4_0-esp-2021r2-patch5 -prefs=runtime.tools.openocd-esp32.path=/Users/lee/Library/Arduino15/packages/esp32/tools/openocd-esp32/v0.11.0-esp32-20220706 -prefs=runtime.tools.openocd-esp32-v0.11.0-esp32-20220706.path=/Users/lee/Library/Arduino15/packages/esp32/tools/openocd-esp32/v0.11.0-esp32-20220706 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc.path=/Users/lee/Library/Arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2021r2-patch5 -prefs=runtime.tools.xtensa-esp32s2-elf-gcc-gcc8_4_0-esp-2021r2-patch5.path=/Users/lee/Library/Arduino15/packages/esp32/tools/xtensa-esp32s2-elf-gcc/gcc8_4_0-esp-2021r2-patch5 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc.path=/Users/lee/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch5 -prefs=runtime.tools.xtensa-esp32s3-elf-gcc-gcc8_4_0-esp-2021r2-patch5.path=/Users/lee/Library/Arduino15/packages/esp32/tools/xtensa-esp32s3-elf-gcc/gcc8_4_0-esp-2021r2-patch5 -prefs=runtime.tools.esptool_py.path=/Users/lee/Library/Arduino15/packages/esp32/tools/esptool_py/4.2.1 -prefs=runtime.tools.esptool_py-4.2.1.path=/Users/lee/Library/Arduino15/packages/esp32/tools/esptool_py/4.2.1 -prefs=runtime.tools.mklittlefs.path=/Users/lee/Library/Arduino15/packages/esp32/tools/mklittlefs/3.0.0-gnu12-dc7f933 -prefs=runtime.tools.mklittlefs-3.0.0-gnu12-dc7f933.path=/Users/lee/Library/Arduino15/packages/esp32/tools/mklittlefs/3.0.0-gnu12-dc7f933 -verbose /Users/lee/Library/Arduino15/packages/esp32/hardware/esp32/2.0.6/libraries/ESP32/examples/ChipID/GetChipID/GetChipID.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/lee/Library/Arduino15/packages -hardware /Users/lee/Documents/Arduino/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/lee/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/lee/Documents/Arduino/libraries -fqbn=esp32:esp32:esp32s3:JTAGAdapter=default,PSRAM=disabled,FlashMode=qio,FlashSize=4M,LoopCore=1,EventsCore=1,USBMode=hwcdc,CDCOnBoot=default,MSCOnBoot=default,DFUOnBoot=default,UploadMode=default,PartitionScheme=default,CPUFreq=240,UploadSpeed=921600,DebugLevel=none,EraseFlash=none -ide-version=10813 -build-path /var/folders/kz/yg8fjhp53c3201sw0bhpccy40000gn/T/arduino_build_129162 -warnings=none -build-cache /var/folders/kz/yg8fjhp53c3201sw0bhpccy40000gn/T/arduino_cache_599257 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=/Users/lee/Library/Arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=/Users/lee/Library/Arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=/Users/lee/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch5 -prefs=runtime.tools.xtensa-esp32-elf-gcc-gcc8_4_0-esp-2021r2-patch5.path=/Users/lee/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2-patch5 -prefs=runtime.tools.riscv32-esp-elf-gcc.path=/Users/lee/Library/Arduino15/packages/esp32/tools/riscv32-esp-elf-gcc/gcc8_4_0-esp-2021r2-patch5 -prefs=runtime.tools.riscv32-esp-elf-gcc-gcc8_4_0-esp-2021r2-patch5.path=/Users/lee/Library/Arduino15/packages/esp32/tools/riscv32-esp-elf-gcc/gcc8_4_0-esp-2021r2-patch5 -prefs=runtime.tools.openocd-esp32.path=/Users/lee/Library/Arduino15/packages/esp32/tools/openocd-esp32/v0.11.0-esp32-20220706 -prefs=runtime.tools.openocd-esp32-v0.11.0-esp32-20220706.path=/Users/l [... cut ...]
[720] Error loading Python lib '/var/folders/kz/yg8fjhp53c3201sw0bhpccy40000gn/T/_MEIHDO9U2/libpython3.8.dylib': dlopen: dlopen(/var/folders/kz/yg8fjhp53c3201sw0bhpccy40000gn/T/_MEIHDO9U2/libpython3.8.dylib, 10): Symbol not found: ____chkstk_darwin
  Referenced from: /var/folders/kz/yg8fjhp53c3201sw0bhpccy40000gn/T/_MEIHDO9U2/libintl.8.dylib (which was built for Mac OS X 11.0)
  Expected in: /usr/lib/libSystem.B.dylib
 in /var/folders/kz/yg8fjhp53c3201sw0bhpccy40000gn/T/_MEIHDO9U2/libintl.8.dylib
exit status 255
/Applications/Arduino.app/Contents/Java/arduino-builder returned 255
Error compiling for board ESP32S3 Dev Module.

mmMicky, 2023-01-31:
It may be caused by an incomplete installation or missing files. Please uninstall ESP32 and reinstall to compile and test.

leetut, 2023-01-31:
Same error, also tried version 205 from boards manager

leetut, 2023-01-31:
Similar issue?
https://github.com/espressif/arduino-esp32/issues/5639

mmMicky, 2023-02-02:
Can you tell me how to fix it? I don't have a mac computer. So don't understand why it's not working. If you are willing to share it, I believe it can help many customers solve this problem.

leetut, 2023-02-02:
esptool is built for macOS 11 minimum, so it fails on 10.13.x

I had to make esptool for my system using this: https://github.com/espressif/arduino-esp32/issues/5639#issuecomment-1023906143

then edit the platform.txt file in: /Users/david/Library/Arduino15/packages/esp32/hardware/esp32/2.0.5

Find the line:

tools.gen_esp32part.cmd=python3 "{runtime.platform.path}/tools/gen_esp32part.py"
Change python3 to the full path given in the "which python3" command earlier. In my case it becomes:

tools.gen_esp32part.cmd=/Library/Frameworks/Python.framework/Versions/3.8/bin/python3 "{runtime.platform.path}/tools/gen_esp32part.py"

lewisxhe, 2023-03-13:
If you still have problems, please open it again

closed 2023-03-13 by lewisxhe