Xinyuan-LilyGO/T-Display-S3 #92: Couldnt upload anythink to Lilygo t-display s3

martinmmi, 2023-01-31:
Hello, iam martin. 

I have the **problem,** that i couldnt upload somethink to to lilygo t-display s3.

I installed everythink simular like this git repository (also the eSPI lib manual in PlatformIO for the tft example) and the first uploads work without problems. 

Now, after connect a BME280 Sensor to SPI/I2C and install BME280 lib from adafruit, i couldnt upload anythink. 

The **ERROR** Message is: A fatal error occurred: Failed to connect to ESP32-S3: *No serial data received*.

What can i do? I tried to put the t-display to bootmode, but i dont know if i did i right. 

Many thanks
-Martin
[image]

mmMicky, 2023-02-01:
You need to manually enter BOOT mode to upload. The operation is to press and hold the BOT button after connecting the USB, and then click the reset button. It doesn't do anything after the operation is complete. The screen stays black.

martinmmi, 2023-02-01:
Thanks Mikky, that helps!

lewisxhe, 2023-03-13:
If you still have problems, please open it again

closed 2023-03-13 by lewisxhe

michelegirolami, 2023-11-29:
dear all, I have a similar problem. I managed to enter in BOOT mode holding the boot button while swithing on the device. After that, the COM port changed and, selecting the new COM port, I managed to upload a custom firmware (very simple). However, after the upload phase is finished, the device does not execute the fw. The screen is black. The COM port is still the one used to upload. 
Is there a way to exist the BOOT mode or this is done right after uploading a fw ? Any suggestion? Thanks

michelegirolami, 2023-12-04:
Dear @teastainGit 
thanks for your message. However, I have a different behavior.  Below my problem:

- I ordered [lilygo ESP32 T3-S3](https://www.lilygo.cc/products/t3s3-v1-0) 
- the device comes with the meshtastic fw
- I managed to enter in the boot mode, by pressing the BOOT button while switching on
- I uploaded a custom fw (e.g. a blink fw) the upload succeed as I can see the upload progress
- after the upload competes, the device does not execute the new fw. The device still remains in boot mode or it does not start the new fw for some reason. Below you can find the output from the serial port:

`ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x4201d546
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0`

I also noticed that if I switch off and on the device WITHOUT pressing the boot botton, then I cannot see the device on the web (no COM port or /dev/tty port assigned)... 
Any idea on how to restore the device so that to execute a custom fw?

teastainGit, 2023-12-04:
Hi [michelegirolami](https://github.com/michelegirolami) I am most familiar with LilyGO T-Display S3 (this repository) and not with meshtastic, which sounds cool.
The standard technique for getting into program transfer mode on LilyGO ESP32 S3 boards is, as you may  know, Hold down Boot, while pressing boot, press and release Reset, then release  boot. in that order. Alternately press and hold boot while plugging in the USB cable. 
If you can see the port, download your meshtastic and when finished you may see (in Arduino IDE ) the message " Leaving...
Hard resetting via RTS pin..."
If the new firmware does not start, try unplugging and plugging in again.
You might try raising an Issue over on your products GitHub repository:
https://github.com/Xinyuan-LilyGO/LilyGo-LoRa-Series/issues
-Terry

svdrummer, 2023-12-04:
I think the word you are looking for is ANYTHIN(G)