Xinyuan-LilyGO/T-Display-S3 #94: How can i use an BME280 I2C Sensor?

martinmmi, 2023-02-01:
Hello, iam martin and new with the t-display-s3.

My Question: How can i use i2c on t-display-s3? In the almost libs you have to assign the pins from the controller. 

I read in the manual from t-display-s3, that i can use every gpio as i2c pins. But how does it work with the pin assignment?

Maybe with the adafruit bme280 lib and Adafruit_Sensor library?
https://randomnerdtutorials.com/bme280-sensor-arduino-pressure-temperature-humidity/

-Martin

teastainGit, 2023-02-01:
I am surprised that the example doesn't include Wire.begin(SDA, SCL); 
 On the LilyGO T-Display S3 the 4pin JST connector beside the USB is 
[image]
SDA pin 43 and SCL pin 44.
So you need to add this to your setup:
Wire.begin(43, 44);  //SDA, SCL
You can use almost any two pins but this is very convenient!
I made a GitHub repository that has a simple example of I2C temp sensor, but a different type, just a demo:
https://github.com/teastainGit/LilyGO-T-display-S3-setup-and-examples
See the HotHead_temp_sensor example.
-Terry

martinmmi, 2023-02-02:
Oh wow. Thanks!! Which advantage has this connector? For me, its realy impractical because i cant can soldering directly and also cant use the standard back arduino pins.

teastainGit, 2023-02-02:
There are several companies that make sensors and boards that use this JST connector. I have drawers full of cool sensors, etc to learn and program, ready to plug in. The LilyGO T-Display S3 use the small JST 1.0 used by Adafruit as "STEMMA QT" and Sparkfun with their compatible QWiiC line of products.
M5Stack has a really good line of controllers and sensors, but they use the larger JST called Grove, also used by the Seeed company.
So, my point is that I have not used a white plastic breadboard for years! 
Worth considering for future hobby projects.
But as you said any two pins can be I2C, you just need to declare them with Wire.begin()

lewisxhe, 2023-03-13:
If you still have problems, please open it again

closed 2023-03-13 by lewisxhe