Xinyuan-LilyGO/T-Display-S3 #95: Buzzer and T-display S3

hammikb, 2023-02-05:
I am currently trying to use a buzzer to set off an alarm when a certain condition is meat but currently it just activates the buzzer and turns the entire display off and the display will flicker back on when the buzzer activates again. Besides that i have to restart it to get the display back on 

```
if (isCelsius) {
      spr2.drawString("A:" + String(alarmtempc,0), 5, 8, 4); 
      if (abs(currentTempc - alarmtempc) <= 1) {
        tone(13, 600, 30);
        noTone(13);
      } else {}   
    } else {
      spr2.drawString("A:" + String(alarmtempf,0), 5, 8, 4);
      if (abs(currentTempf- alarmtempf) <= 1) {
        tone(13, 600, 30);
        noTone(13);
      } else {} 
    }
```

svdrummer, 2023-02-05:
We can assume you have set the correct settings in the SETUP section.
Seems like the current draw may be too much on the buzzer. Try running without anything connected to pin I/O13
Could be a brownout

hammikb, 2023-02-05:
if my mean correct setting in the SETUP by what is below then yes, if not please let me know. 
`pinMode(13, OUTPUT); `

it actually happens without the buzzer attached to the pins, do you know would i would fix this?

pdurys, 2023-02-25:
I guess you have tried to setup different gpio for buzzer instead of #13.

svdrummer, 2023-02-25:
We are assuming your buzzer has worked before. Two types of buzzer, passive and active. If passive, just putting power to it will note make a noise. See Arduino tone.
Put power in the battery and see if that makes a noise.

lewisxhe, 2023-03-13:
If you still have problems, please open it again

closed 2023-03-13 by lewisxhe