Xinyuan-LilyGO/T-Display-S3 #98: Library don't work in Arduino IDE (2.0.3)

laverdanny, 2023-02-05:
and example/factory/factory.ino need unknown #include "TouchLib.h" (several are possible, maybe a special one ?)

teastainGit, 2023-02-05:
and it's here:
https://github.com/Xinyuan-LilyGO/T-Display-S3/tree/main/lib/TouchLib/src

laverdanny, 2023-02-05:
I see, thanks !
any way to correctly install [T-Display-S3] library in Arduino ?
 several examples fails to compile and/or run once downloaded

And choosing communication port, Arduino IDE shows ESP32C3 Dev module instead of ESP32S3

teastainGit, 2023-02-05:
I had a lot of trouble getting the LilyGO T-Display S3 to work, so I made a repository that just kept getting bigger and bigger.
https://github.com/teastainGit/LilyGO-T-display-S3-setup-and-examples
I say bigger because it is a lot of reading, but it will get you going with stable examples...you can branch out from there.
Arduino V2.0.3 is very good and it now seems to save the upload settings but it will switch boards on you sometimes. Just a bug, I suppose.
-Terry

laverdanny, 2023-02-05:
Actually you're right, board name keep changing !
Thanks ! going to read a lot ;-)

Felixrising, 2023-02-09:
Really the examples need reworking. Thanks teastainGit for your repo. Lots of issues with the board definition and libraries/examples not working or being a bit too brittle to reliably use. This repo needs rework! Quite a frustrating experience for a novice. :/

lewisxhe, 2023-03-13:
If you still have problems, please restart

closed 2023-03-13 by lewisxhe