Xinyuan-LilyGO/T-Display-S3 #99: Memory partitioning

laverdanny, 2023-02-06:
hi there
As far as I know Lilygo suggest to choose partition scheme  "Huge APP (3MB No OTA/1MB SPIFFS)" for  T-Display-S3

Actually this device is 16MB equipped.

I saw a partition 16MB (3MB No OTA/9,9MB FATFS), is there a way to set larger areas to APP and/or SPIFFS ?

Regards

mmMicky, 2023-02-07:
The use of "Huge APP (3MB No OTA/1MB SPIFFS)" in the routine is for the sake of versatility. If you need to use more file system space, you need to modify the .csv partition table configuration file. Cannot use Arduino IDE. It cannot specify a user's .csv file. Only PlatformIO or ESP-IDF can be used.

lewisxhe, 2023-03-13:
If you still have problems, please reopen it

closed 2023-03-13 by lewisxhe