Xinyuan-LilyGO/T-Display-S3 #100: Upload from PlatformIO on an Acer laptop.

wiegleyj, 2023-02-07:
When I try to upload to a T-Display S3 It fails with the following:

```
 *  Executing task in folder QuickTestWithESP: C:\Users\Sanam\.platformio\penv\Scripts\platformio.exe run --target upload 

Processing esp32-s3-devkitc-1 (platform: espressif32; board: esp32-s3-devkitc-1; framework: espidf)
--------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitc-1.html
PLATFORM: Espressif 32 (6.0.0) > Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf @ 3.50000.0 (5.0.0)
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.40400.0 (4.4.0)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-mkfatfs @ 2.0.1
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-mkspiffs @ 2.230.0 (2.30)
 - tool-ninja @ 1.9.0
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0)
 - toolchain-riscv32-esp @ 11.2.0+2022r1
 - toolchain-xtensa-esp32s3 @ 11.2.0+2022r1
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Retrieving maximum program size .pio\build\esp32-s3-devkitc-1\firmware.elf
Checking size .pio\build\esp32-s3-devkitc-1\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   4.1% (used 13504 bytes from 327680 bytes)
Flash: [==        ]  19.6% (used 205693 bytes from 1048576 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-builtin, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: COM3
Uploading .pio\build\esp32-s3-devkitc-1\firmware.bin
esptool.py v4.4
Serial port COM3
Connecting...
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 68:b6:b3:22:d9:28
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x00010000 to 0x00042fff...
Compressed 20816 bytes to 13278...
Writing at 0x00000000... (100 %)
Wrote 20816 bytes (13278 compressed) at 0x00000000 in 0.4 seconds (effective 423.3 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 103...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.1 seconds (effective 388.4 
kbit/s)...
Hash of data verified.
Compressed 206080 bytes to 107219...
Writing at 0x00010000... (14 %)
Writing at 0x0001c421... (28 %)
Traceback (most recent call last):
  File "C:\Users\Sanam\.platformio\packages\tool-esptoolpy\esptool.py", line 34, in <module>
    esptool._main()
  File "C:\Users\Sanam\.platformio\packages\tool-esptoolpy\esptool\__init__.py", line 1026, in _main
    main()
  File "C:\Users\Sanam\.platformio\packages\tool-esptoolpy\esptool\__init__.py", line 826, in main
    operation_func(esp, args)
  File "C:\Users\Sanam\.platformio\packages\tool-esptoolpy\esptool\cmds.py", line 556, in write_flash
    esp.flash_defl_block(block, seq, timeout=timeout)
  File "C:\Users\Sanam\.platformio\packages\tool-esptoolpy\esptool\loader.py", line 109, in inner
    return func(*args, **kwargs)
  File "C:\Users\Sanam\.platformio\packages\tool-esptoolpy\esptool\loader.py", line 991, in flash_defl_block
    self.check_command(
  File "C:\Users\Sanam\.platformio\packages\tool-esptoolpy\esptool\loader.py", line 407, in check_command
    val, data = self.command(op, data, chk, timeout=timeout)
  File "C:\Users\Sanam\.platformio\packages\tool-esptoolpy\esptool\loader.py", line 376, in command
    p = self.read()
  File "C:\Users\Sanam\.platformio\packages\tool-esptoolpy\esptool\loader.py", line 308, in read
    return next(self._slip_reader)
StopIteration
*** [upload] Error 1
========================= [FAILED] Took 41.92 seconds =========================

 *  The terminal process "C:\Users\Sanam\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'upload'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

It should be working. I have tested the exact same setup project with a official Espressif ESP32-S3-DevkitC-1 and it works/uploads just fine.

```
 *  The terminal process "C:\Users\Sanam\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'upload'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 

 *  Executing task in folder QuickTestWithESP: C:\Users\Sanam\.platformio\penv\Scripts\platformio.exe run --target upload 

Processing esp32-s3-devkitc-1 (platform: espressif32; board: esp32-s3-devkitc-1; framework: espidf)
--------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitc-1.html
PLATFORM: Espressif 32 (6.0.0) > Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf @ 3.50000.0 (5.0.0)
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.40400.0 (4.4.0)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-mkfatfs @ 2.0.1
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-mkspiffs @ 2.230.0 (2.30)
 - tool-ninja @ 1.9.0
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0)
 - toolchain-riscv32-esp @ 11.2.0+2022r1
 - toolchain-xtensa-esp32s3 @ 11.2.0+2022r1
Reading CMake configuration...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Retrieving maximum program size .pio\build\esp32-s3-devkitc-1\firmware.elf
Checking size .pio\build\esp32-s3-devkitc-1\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   4.1% (used 13504 bytes from 327680 bytes)
Flash: [==        ]  19.6% (used 205693 bytes from 1048576 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-builtin, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: COM6
Uploading .pio\build\esp32-s3-devkitc-1\firmware.bin
esptool.py v4.4
Serial port COM6
Connecting....
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 7c:df:a1:e8:7f:dc
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00000000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x00010000 to 0x00042fff...
Compressed 20816 bytes to 13278...
Writing at 0x00000000... (100 %)
Wrote 20816 bytes (13278 compressed) at 0x00000000 in 0.7 seconds (effective 232.2 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 103...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.1 seconds (effective 337.3 
kbit/s)...
Hash of data verified.
Compressed 206080 bytes to 107219...
Writing at 0x00010000... (14 %)
Writing at 0x0001c421... (28 %)
Writing at 0x0002213a... (42 %)
Writing at 0x00027c31... (57 %)
Writing at 0x000309d9... (71 %)
Writing at 0x00038d9c... (85 %)
Writing at 0x0003ef8b... (100 %)
Wrote 206080 bytes (107219 compressed) at 0x00010000 in 3.1 seconds (effective 525.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
========================= [SUCCESS] Took 43.23 seconds ========================= *  Terminal will be reused by tasks, press any key to close it. 
```

And platformio.ini might be helpful:
```
[env:esp32-s3-devkitc-1]
platform = espressif32
board = esp32-s3-devkitc-1
framework = espidf
board_build.flash_mode = dio
```
Please, any ideas how to fix this.

mmMicky, 2023-02-08:
Enter boot mode manually. The operation is to hold down the BOT button and then click the RST button.

tyeth, 2023-02-08:
I've had problems related to timeouts, you can change the esptool config to timeout more slowly, but to be honest it's usually a sign of a poor cable, or having peripherals attached while flashing (usb power on laptop not enough). I was tracking [this issue](https://github.com/espressif/esptool/issues/793), and I modified my loader.py file, but this is the new help page about using a custom config file to modify timeouts so you don't modify any esptool script files and instead create a user override config file: https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/configuration-file.html
I'm not helping myself, by having libUSBcap (packet capture for usb) running which increases timeout likelyhood, plus peripherals attached. I basically doubled or quadrupled some of the timings and that increased my success rate to 100% for small firmwares or above 80% for the larger 2MB firmwares.

pdurys, 2023-02-08:
I have the same issue, and solution to enter boot manually makes the trick. In my situation I guess serial port was spammed by Serial.print() from running sketch. It seems that Arduino IDE is more stable in this matter than platformio.

lewisxhe, 2023-03-13:
If you still have problems, please reopen it

closed 2023-03-13 by lewisxhe