Xinyuan-LilyGO/T-Display-S3 #101: GPIO 0 always on

pdurys, 2023-02-08:
I have very simple sketch having only 2 external libraries (see my platforio.ini file)

`
[env:esp32-s3-devkitc-1]
platform = espressif32
board = esp32-s3-devkitc-1
framework = arduino
platform_packages = 
	framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#2.0.5
build_flags = 
	-D LV_LVGL_H_INCLUDE_SIMPLE
	-D BOARD_HAS_PSRAM
	-D ARDUINO_USB_MODE=1
	-D ARDUINO_USB_CDC_ON_BOOT=1
;board_build.partitions = default_8MB.csv
board_build.arduino.memory_type = qio_opi
;board_build.flash_size = 8MB
monitor_speed = 115200
lib_deps = 
	mathertel/OneButton@^2.0.3
   	esphome/ESP32-audioI2S@^2.0.6
`

It simply create Audio object but even do not play stream.
Another bit of sketch is simple monitor and Serial.println() events when onboard buttons at GPIO 0 and 14 are pressed, double click or longpressed. See below.

`#include <Arduino.h>

#include <Audio.h> // https://github.com/schreibfaul1/ESP32-audioI2S

// MAX98357A I2S pins.
#define I2S_WS 10
#define I2S_BCK 11
#define I2S_DOUT 12

Audio audio;

#include "OneButton.h" // https://github.com/mathertel/OneButton

#define PIN_BUTTON_1   0
#define PIN_BUTTON_2  14
#define PIN_BAT_VOLT   4

// Setup a new OneButtons
OneButton button1(PIN_BUTTON_1, true);
OneButton button2(PIN_BUTTON_2, true);

// ----- button 1 callback functions

// This function will be called when the button1 was pressed 1 time (and no 2. button press followed).
void click1()
{
  Serial.println("Button 1 click.");
} // click1

// This function will be called when the button1 was pressed 2 times in a short timeframe.
void doubleclick1()
{
  Serial.println("Button 1 doubleclick.");
} // doubleclick1

// This function will be called once, when the button1 is pressed for a long time.
void longPressStart1()
{
  Serial.println("Button 1 longPress start");
} // longPressStart1

// This function will be called often, while the button1 is pressed for a long time.
void longPress1()
{
  Serial.println("Button 1 longPress...");
  delay(2000);
} // longPress1

// This function will be called once, when the button1 is released after beeing pressed for a long time.
void longPressStop1()
{
  Serial.println("Button 1 longPress stop");
} // longPressStop1

// ... and the same for button 2:

void click2()
{
  Serial.println("Button 2 click.");
} // click2

void doubleclick2()
{
  Serial.println("Button 2 doubleclick.");
} // doubleclick2

void longPressStart2()
{
  Serial.println("Button 2 longPress start");
} // longPressStart2

void longPress2()
{
  Serial.println("Button 2 longPress...");
    delay(2000);
} // longPress2

void longPressStop2()
{
  Serial.println("Button 2 longPress stop");
} // longPressStop2

void setup() {

  delay(5000);

  Serial.begin(115200);
  Serial.println("Hello T-Display-S3");
  delay(5000);

  // Select I2S pins
  audio.setPinout(I2S_BCK, I2S_WS, I2S_DOUT);

  // Volume control.
  audio.setVolume(21);

  // Buttons
  Serial.println("Starting TwoButtons...");
  delay(10000);

  // link the button 1 functions.
  button1.attachClick(click1);
  button1.attachDoubleClick(doubleclick1);
  button1.attachLongPressStart(longPressStart1);
  button1.attachLongPressStop(longPressStop1);
  button1.attachDuringLongPress(longPress1);

  // link the button 2 functions.
  button2.attachClick(click2);
  button2.attachDoubleClick(doubleclick2);
  button2.attachLongPressStart(longPressStart2);
  button2.attachLongPressStop(longPressStop2);
  button2.attachDuringLongPress(longPress2);
}

void loop() {

  // Start the stream.
  audio.loop();

  // Keep watching the push buttons
  button1.tick();
  button2.tick();

}
`

When I execute this there is constant 
```
Button 1 longPress
```
 event. Of course you can see also 
```
Button 1 longPress start 
```
at the begin but later there is only Button 1 longPress. Like it would be on GPIO constant LOW signal.

This problem exists while building on platformio, when doing exactly the same on Arduino IDE all is just fine.

teastainGit, 2023-02-08:
I see you are using arduino-esp32.git#2.0.5?
I had trouble with that rev with some boards!
The latest is:
 https://github.com/espressif/arduino-esp32#2.0.6
cheers, Terry

pdurys, 2023-02-08:
Unfortunately 2.0.6 is nothing better in this case.
I guess you have that board. Could you check if you have the same effect? 
cheers, pdurys

teastainGit, 2023-02-08:
Sorry!
I use Arduino IDE V2.0.3 and as you say, it works.

pdurys, 2023-02-13:
Update.
I have managed to edit original factory example, adding Audio library etc.
It works nicely without problem.

But, there is always catch somewhere.

I have had an idea to copy all src files from Arduino project tree to PlatformIO tree. So far so good.
And problem has return.

This mean I have used exactly the same source files from modified script which correctly compiles and uploads inside PIO located inside original Arduino tree, but when open folder in PIO located within PIO tree and use those files, I have the same problem as before. GPIO0 seems to be always LOW.
It must be something with either lvgl library included in LilyGo T-Display-S3 repository vs library from PIO or something else.

pdurys, 2023-02-14:
I think I have found explanation with this reported Issue: https://github.com/schreibfaul1/ESP32-audioI2S/issues/408#issue-1438366058

audioI2S library was setting GPIO0 for MCLK by default. Latest library version seems to resolve it.

closed 2023-02-14 by pdurys