Xinyuan-LilyGO/T-Display-S3 #103: T-Display-S3 and ESP-IDF v5.0

KamranAghlami, 2023-02-15:
Hey everyone!

I've been pulling my hair out in few past days trying to get the display working, I have done it in IDF v4.4 (https://github.com/KamranAghlami/T-Display-S3.git) but for some reason I can't figure it out in version 5.0, I'm a software guy with relatively limited knowledge of hardware.

Thought I'd ask here if any one has done it or has any idea why my code isn't working.
I'll copy my example code which is based on esp32's i80 display example.

Thanks in advance

``` C
#include <stdio.h>
#include "string.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_timer.h"
#include "esp_lcd_panel_io.h"
#include "esp_lcd_panel_vendor.h"
#include "esp_lcd_panel_ops.h"
#include "driver/gpio.h"
#include "esp_err.h"
#include "esp_log.h"

static const char *TAG = "example";

////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
//////////////////// Please update the following configuration according to your LCD spec //////////////////////////////
////////////////////////////////////////////////////////////////////////////////////////////////////////////////////////
#define EXAMPLE_LCD_PIXEL_CLOCK_HZ (2 * 1000 * 1000)

#define EXAMPLE_LCD_BK_LIGHT_ON_LEVEL 1
#define EXAMPLE_LCD_BK_LIGHT_OFF_LEVEL !EXAMPLE_LCD_BK_LIGHT_ON_LEVEL
#define EXAMPLE_PIN_NUM_DATA0 39    // 6
#define EXAMPLE_PIN_NUM_DATA1 40    // 7
#define EXAMPLE_PIN_NUM_DATA2 41    // 8
#define EXAMPLE_PIN_NUM_DATA3 42    // 9
#define EXAMPLE_PIN_NUM_DATA4 45    // 10
#define EXAMPLE_PIN_NUM_DATA5 46    // 11
#define EXAMPLE_PIN_NUM_DATA6 47    // 12
#define EXAMPLE_PIN_NUM_DATA7 48    // 13
#define EXAMPLE_PIN_NUM_PCLK 8      // 5
#define EXAMPLE_PIN_NUM_CS 6        // 3
#define EXAMPLE_PIN_NUM_DC 7        // 4
#define EXAMPLE_PIN_NUM_RST 5       // 2
#define EXAMPLE_PIN_NUM_BK_LIGHT 38 // 1
#define EXAMPLE_PIN_NUM_POWER 15

// The pixel number in horizontal and vertical
#define EXAMPLE_LCD_H_RES 320
#define EXAMPLE_LCD_V_RES 170
// Bit number used to represent command and parameter
#define EXAMPLE_LCD_CMD_BITS 8
#define EXAMPLE_LCD_PARAM_BITS 8

#define EXAMPLE_LVGL_TICK_PERIOD_MS 2

// Supported alignment: 16, 32, 64. A higher alignment can enables higher burst transfer size, thus a higher i80 bus throughput.
#define EXAMPLE_PSRAM_DATA_ALIGNMENT 32

void *buf1 = NULL;
void *buf2 = NULL;
size_t buff_idx = 0;
static bool trans_done = true;

static void write_color(esp_lcd_panel_handle_t panel_handle)
{
    if (!trans_done)
        return;

    trans_done = false;

    void *buff = (buff_idx++ % 2) ? buf1 : buf2;

    esp_lcd_panel_draw_bitmap(panel_handle, 32, 32, 64, 64, buff);
}

size_t trans_done_calls = 0;

static bool on_color_trans_done(esp_lcd_panel_io_handle_t panel_io, esp_lcd_panel_io_event_data_t *edata, void *user_ctx)
{
    trans_done = true;
    trans_done_calls++;

    return false;
}

static void tick_timer_cb(void *arg)
{
    ESP_LOGI(TAG, "trans_done_calls: %zu", trans_done_calls);

    trans_done_calls = 0;
}

void app_main(void)
{
    gpio_config_t pwr_gpio_config = {
        .mode = GPIO_MODE_OUTPUT,
        .pin_bit_mask = 1ULL << EXAMPLE_PIN_NUM_POWER};
    ESP_ERROR_CHECK(gpio_config(&pwr_gpio_config));
    gpio_set_level(EXAMPLE_PIN_NUM_POWER, EXAMPLE_LCD_BK_LIGHT_ON_LEVEL);

    ESP_LOGI(TAG, "Turn off LCD backlight");
    gpio_config_t bk_gpio_config = {
        .mode = GPIO_MODE_OUTPUT,
        .pin_bit_mask = 1ULL << EXAMPLE_PIN_NUM_BK_LIGHT};
    ESP_ERROR_CHECK(gpio_config(&bk_gpio_config));
    gpio_set_level(EXAMPLE_PIN_NUM_BK_LIGHT, EXAMPLE_LCD_BK_LIGHT_OFF_LEVEL);

    ESP_LOGI(TAG, "Initialize Intel 8080 bus");
    esp_lcd_i80_bus_handle_t i80_bus = NULL;
    esp_lcd_i80_bus_config_t bus_config = {
        .clk_src = LCD_CLK_SRC_DEFAULT,
        .dc_gpio_num = EXAMPLE_PIN_NUM_DC,
        .wr_gpio_num = EXAMPLE_PIN_NUM_PCLK,
        .data_gpio_nums = {
            EXAMPLE_PIN_NUM_DATA0,
            EXAMPLE_PIN_NUM_DATA1,
            EXAMPLE_PIN_NUM_DATA2,
            EXAMPLE_PIN_NUM_DATA3,
            EXAMPLE_PIN_NUM_DATA4,
            EXAMPLE_PIN_NUM_DATA5,
            EXAMPLE_PIN_NUM_DATA6,
            EXAMPLE_PIN_NUM_DATA7,
        },
        .bus_width = CONFIG_EXAMPLE_LCD_I80_BUS_WIDTH,
        .max_transfer_bytes = EXAMPLE_LCD_H_RES * 100 * sizeof(uint16_t),
        .psram_trans_align = EXAMPLE_PSRAM_DATA_ALIGNMENT,
        .sram_trans_align = 4,
    };
    ESP_ERROR_CHECK(esp_lcd_new_i80_bus(&bus_config, &i80_bus));
    esp_lcd_panel_io_handle_t io_handle = NULL;
    esp_lcd_panel_io_i80_config_t io_config = {
        .cs_gpio_num = EXAMPLE_PIN_NUM_CS,
        .pclk_hz = EXAMPLE_LCD_PIXEL_CLOCK_HZ,
        .trans_queue_depth = 20,
        .dc_levels = {
            .dc_idle_level = 0,
            .dc_cmd_level = 0,
            .dc_dummy_level = 0,
            .dc_data_level = 1,
        },
        .flags = {
            .swap_color_bytes = 1, // Swap can be done in LvGL (default) or DMA
        },
        .on_color_trans_done = on_color_trans_done,
        .user_ctx = NULL,
        .lcd_cmd_bits = EXAMPLE_LCD_CMD_BITS,
        .lcd_param_bits = EXAMPLE_LCD_PARAM_BITS,
    };
    ESP_ERROR_CHECK(esp_lcd_new_panel_io_i80(i80_bus, &io_config, &io_handle));

    esp_lcd_panel_handle_t panel_handle = NULL;

    ESP_LOGI(TAG, "Install LCD driver of st7789");
    esp_lcd_panel_dev_config_t panel_config = {
        .reset_gpio_num = EXAMPLE_PIN_NUM_RST,
        .rgb_endian = LCD_RGB_ENDIAN_RGB,
        .bits_per_pixel = 16,
    };
    ESP_ERROR_CHECK(esp_lcd_new_panel_st7789(io_handle, &panel_config, &panel_handle));

    esp_lcd_panel_reset(panel_handle);
    esp_lcd_panel_init(panel_handle);
    // Set inversion, x/y coordinate order, x/y mirror according to your LCD module spec
    // the gap is LCD panel specific, even panels with the same driver IC, can have different gap value
    ESP_ERROR_CHECK(esp_lcd_panel_invert_color(panel_handle, true));
    ESP_ERROR_CHECK(esp_lcd_panel_swap_xy(panel_handle, true));
    ESP_ERROR_CHECK(esp_lcd_panel_mirror(panel_handle, false, true));
    ESP_ERROR_CHECK(esp_lcd_panel_set_gap(panel_handle, 0, 35));

    ESP_LOGI(TAG, "Turn on LCD backlight");
    gpio_set_level(EXAMPLE_PIN_NUM_BK_LIGHT, EXAMPLE_LCD_BK_LIGHT_ON_LEVEL);

    ESP_LOGI(TAG, "Initialize UI");
    // alloc draw buffers used by LVGL
    // it's recommended to choose the size of the draw buffer(s) to be at least 1/10 screen sized
    buf1 = heap_caps_malloc(EXAMPLE_LCD_H_RES * 100 * 2, MALLOC_CAP_DMA | MALLOC_CAP_INTERNAL);
    buf2 = heap_caps_malloc(EXAMPLE_LCD_H_RES * 100 * 2, MALLOC_CAP_DMA | MALLOC_CAP_INTERNAL);

    assert(buf1);
    assert(buf2);

    memset(buf1, 0x0, EXAMPLE_LCD_H_RES * 100 * 2);
    memset(buf2, 0xff, EXAMPLE_LCD_H_RES * 100 * 2);

    ESP_LOGI(TAG, "buf1@%p, buf2@%p", buf1, buf2);

    const esp_timer_create_args_t tick_timer_args = {
        .callback = &tick_timer_cb,
        .name = "tick_timer"};
    esp_timer_handle_t tick_timer = NULL;
    ESP_ERROR_CHECK(esp_timer_create(&tick_timer_args, &tick_timer));
    ESP_ERROR_CHECK(esp_timer_start_periodic(tick_timer, 1000000));

    // user can flush pre-defined pattern to the screen before we turn on the screen or backlight
    ESP_ERROR_CHECK(esp_lcd_panel_disp_on_off(panel_handle, true));

    while (1)
    {
        write_color(panel_handle);

        vTaskDelay(pdMS_TO_TICKS(10));
    }
}
```

mmMicky, 2023-02-16:
Can you attach the output log information? It allows us to combine code judgments to find problems.

KamranAghlami, 2023-02-16:
> Can you attach the output log information? It allows us to combine code judgments to find problems.

Sure, here's the output I'm getting:

``` 
-- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM6 instead...
--- idf_monitor on \\.\COM6 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xa (SPI_FAST_FLASH_BOOT)
Saved PC:0x40378f3e
0x40378f3e: esp_cpu_wait_for_intr at C:/Users/Kamran/.espressif/esp-idf/components/esp_hw_support/cpu.c:110

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x17d8
load:0x403c9700,len:0xe88
load:0x403cc700,len:0x3000
SHA-256 comparison failed:
Calculated: c89dc02e723f073e74738979af16be3c235a46d7f326107e015929d30b46570b
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff  
Attempting to boot anyway...
entry 0x403c9930
I (43) boot: ESP-IDF v5.0 2nd stage bootloader
I (43) boot: compile time 15:45:01
I (43) boot: chip revision: v0.1
I (45) boot_comm: chip revision: 1, min. bootloader chip revision: 0        
I (52) qio_mode: Enabling QIO for flash chip WinBond
I (57) boot.esp32s3: Boot SPI Speed : 80MHz
I (62) boot.esp32s3: SPI Mode       : QIO
I (67) boot.esp32s3: SPI Flash Size : 16MB       
I (72) boot: Enabling RNG early entropy source...
I (77) boot: Partition Table:
I (81) boot: ## Label            Usage          Type ST Offset   Length  
I (88) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (95) boot:  1 phy_init         RF data          01 01 0000f000 00001000 
I (103) boot:  2 factory          factory app      00 00 00010000 00100000
I (110) boot: End of partition table
I (115) boot_comm: chip revision: 1, min. application chip revision: 0
I (122) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=0a68ch ( 42636) map
I (135) esp_image: segment 1: paddr=0001a6b4 vaddr=3fc92c00 size=03258h ( 12888) load
I (141) esp_image: segment 2: paddr=0001d914 vaddr=40374000 size=02704h (  9988) load
I (149) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=1d1e4h (119268) map
I (168) esp_image: segment 4: paddr=0003d20c vaddr=40376704 size=0c400h ( 50176) load
I (176) esp_image: segment 5: paddr=00049614 vaddr=50000000 size=00010h (    16) load
I (182) boot: Loaded app from partition at offset 0x10000
I (182) boot: Disabling RNG early entropy source...    
I (197) octal_psram: vendor id    : 0x0d (AP)
I (197) octal_psram: dev id       : 0x02 (generation 3)
I (197) octal_psram: density      : 0x03 (64 Mbit)     
I (202) octal_psram: good-die     : 0x01 (Pass)        
I (207) octal_psram: Latency      : 0x01 (Fixed)       
I (212) octal_psram: VCC          : 0x01 (3V)
I (217) octal_psram: SRF          : 0x01 (Fast Refresh)
I (223) octal_psram: BurstType    : 0x01 (Hybrid Wrap)    
I (229) octal_psram: BurstLen     : 0x01 (32 Byte)        
I (235) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (241) octal_psram: DriveStrength: 0x00 (1/1)
W (246) PSRAM: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.
I (257) esp_psram: Found 8MB PSRAM device
I (261) esp_psram: Speed: 80MHz
I (265) cpu_start: Pro cpu up.
I (268) cpu_start: Starting app cpu, entry point is 0x40375398
0x40375398: call_start_cpu1 at C:/Users/Kamran/.espressif/esp-idf/components/esp_system/port/cpu_start.c:142

I (0) cpu_start: App cpu up.
I (720) esp_psram: SPI SRAM memory test OK
I (729) cpu_start: Pro cpu start user code
I (729) cpu_start: cpu freq: 240000000 Hz
I (729) cpu_start: Application information:
I (732) cpu_start: Project name:     display_driver
I (738) cpu_start: App version:      86b7fb2
I (743) cpu_start: Compile time:     Feb 16 2023 12:55:55
I (749) cpu_start: ELF file SHA256:  7409bb4d5900cd7f...
I (755) cpu_start: ESP-IDF:          v5.0
I (760) heap_init: Initializing. RAM available for dynamic allocation:
I (767) heap_init: At 3FC968E0 len 00052E30 (331 KiB): D/IRAM
I (773) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (780) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (786) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM
I (792) esp_psram: Adding pool of 8192K of PSRAM memory to heap allocator
I (800) spi_flash: detected chip: winbond
I (805) spi_flash: flash io: qio
I (809) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (829) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (829) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (839) example: Turn off LCD backlight
I (849) gpio: GPIO[38]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (859) example: Initialize Intel 8080 bus
I (859) example: Install LCD driver of st7789
I (869) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (999) example: Turn on LCD backlight
I (999) example: Initialize UI
I (999) example: buf1@0x3fc9f388, buf2@0x3fcaed8c
I (1999) example: trans_done_calls: 101
I (2999) example: trans_done_calls: 100
I (3999) example: trans_done_calls: 100
I (4999) example: trans_done_calls: 100
I (5999) example: trans_done_calls: 100
I (6999) example: trans_done_calls: 100
I (7999) example: trans_done_calls: 100
I (8999) example: trans_done_calls: 100
I (9999) example: trans_done_calls: 100
I (10999) example: trans_done_calls: 100
I (11999) example: trans_done_calls: 100
I (12999) example: trans_done_calls: 100
I (13999) example: trans_done_calls: 100
I (14999) example: trans_done_calls: 100
```

mmMicky, 2023-02-17:
I read carefully several times, but couldn't find the crux of the problem. I'm going to tag it, hope someone in the know can help.

KamranAghlami, 2023-02-17:
Thanks for spending time on this, I'm looking forward for some help as well.

NhiepPhongPTT, 2023-03-02:
I have the same error as you. I fixed it by copying the entire esp_lcd folder in the components of ESP-IDF v4.4.4 to the corresponding path of ESP-IDF v5.0.1 (../esp-idf/components/esp_lcd)

KamranAghlami, 2023-03-04:
hmmm, It's nice to know doing that fixes the issue, but it's far from ideal to revert to an old version of the SDK specially when it's a major one. It's a good starting point to figure out what has changed to fix the actual issue though. 👍

NhiepPhongPTT, 2023-03-04:
You can give feedback to the ESP-IDF team so they can fix it in a later version while we can use the older version. The ESP-IDF version is still the best version with only one component in it being old. If you want to find out specifically where the error is, just check the programs related to initializing the lcd, this is not difficult but it will take time.

krupis, 2023-04-25:
Has there been any updates regarding this? I was just about to purchase this display and I want to use it with esp-idf v5

krupis, 2023-04-25:
> > Has there been any updates regarding this? I was just about to purchase this display and I want to use it with esp-idf v5
> 
> No, unfortunately I have yet to find spare time to work on this. any further help is appreciated.

Thanks for the quick response. Have you posted feature request on esp-idf git yet

KamranAghlami, 2023-04-25:
> Thanks for the quick response. Have you posted feature request on esp-idf git yet?

Sure, It's something that I'd like to see resolved myself.
Regarding your question, not really, since I can't confirm it's an IDF issue in the first place.

minhnt97, 2023-04-25:
@KamranAghlami Could you show me the OUTPUT LOG part when the program crashed and got rebooted? The part with Guru Meditation Error... I believe I have the same issue although I used another I2C sensor in my case

KamranAghlami, 2023-04-25:
> @KamranAghlami Could you show me the OUTPUT LOG part when the program crashed and got rebooted? The part with Guru Meditation Error... I believe I have the same issue although I used another I2C sensor in my case

I'm not sure which crash/reboot you are referring to, my sample never crashes

KamranAghlami, 2023-04-25:
Weird thing is if I touch sides of the board some times I get an screen refresh with the correct color but wrong shape and at wrong position, I'd say It's a signal corruption but It's how and why is beyond me.

simonchatts, 2023-05-15:
Thanks @hoffmajs - that solved the issue for me.

KamranAghlami, 2023-05-16:
Thanks @hoffmajs, that was it, can't believe I missed that. 🤦‍♂️ 👍

closed 2023-05-16 by KamranAghlami

krupis, 2023-05-23:
Hello. Sorry for reopening this closed issue but I would like your help getting the T-Display S3 to work. 
What I did:

1. Open the esp\esp-idf\examples\peripherals\lcd\i80_controller example project
2. Replace all the code with the your code in your first comment
3. In the menu config, the following example settings are selected:
[image]
4, Added 

    ```const gpio_config_t input_conf = {
    .mode = GPIO_MODE_INPUT,
    .pull_up_en = GPIO_PULLUP_ENABLE,
    .pin_bit_mask =  1ULL << PIN_LCD_RD
    };
    ESP_ERROR_CHECK(gpio_config(&input_conf));
    ```
    In app main.
5. Succesfully built and flashed the T-Display S3 (without touch) module

The device is constantly rebooting due to failed to init external RAM:
```
ELF file SHA256: c83a81ada7bdb9e7

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x9 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375b04
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x164c
load:0x403c9700,len:0xbe0
load:0x403cc700,len:0x2ef8
entry 0x403c9900
[1B][0;32mI (24) boot: ESP-IDF v5.0.1-dirty 2nd stage bootloader[1B][0m
[1B][0;32mI (24) boot: compile time 08:05:20[1B][0m
[1B][0;32mI (24) boot: chip revision: v0.1[1B][0m
[1B][0;32mI (26) boot.esp32s3: Boot SPI Speed : 80MHz[1B][0m
[1B][0;32mI (31) boot.esp32s3: SPI Mode       : DIO[1B][0m
[1B][0;32mI (36) boot.esp32s3: SPI Flash Size : 2MB[1B][0m
[1B][0;32mI (41) boot: Enabling RNG early entropy source...[1B][0m
[1B][0;32mI (46) boot: Partition Table:[1B][0m
[1B][0;32mI (50) boot: ## Label            Usage          Type ST Offset   Length[1B][0m
[1B][0;32mI (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000[1B][0m
[1B][0;32mI (64) boot:  1 phy_init         RF data          01 01 0000f000 00001000[1B][0m
[1B][0;32mI (72) boot:  2 factory          factory app      00 00 00010000 00100000[1B][0m
[1B][0;32mI (79) boot: End of partition table[1B][0m
[1B][0;32mI (84) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=0af9ch ( 44956) map[1B][0m
[1B][0;32mI (100) esp_image: segment 1: paddr=0001afc4 vaddr=3fc92100 size=0277ch ( 10108) load[1B][0m
[1B][0;32mI (103) esp_image: segment 2: paddr=0001d748 vaddr=40374000 size=028d0h ( 10448) load[1B][0m
[1B][0;32mI (111) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=1d72ch (120620) map[1B][0m
[1B][0;32mI (139) esp_image: segment 4: paddr=0003d754 vaddr=403768d0 size=0b768h ( 46952) load[1B][0m
[1B][0;32mI (155) boot: Loaded app from partition at offset 0x10000[1B][0m
[1B][0;32mI (156) boot: Disabling RNG early entropy source...[1B][0m
[1B][0;31mE (167) quad_psram: PSRAM ID read error: 0x00ffffff[1B][0m
[1B][0;31mE (167) cpu_start: Failed to init external RAM![1B][0m

abort() was called at PC 0x40375429 on core 0

Backtrace: 0x40375e96:0x3fceb220 0x4037a9bd:0x3fceb240 0x4037ffea:0x3fceb260 0x40375429:0x3fceb2d0 0x403cd8ea:0x3fceb330 0x403cdcf1:0x3fceb380 0x403c9969:0x3fceb4b0 0x40045c01:0x3fceb570 |<-CORRUPTED

ELF file SHA256: c83a81ada7bdb9e7
```

Is your example not running on T-Display S3 board? Do you have your custom board with external ram?

krupis, 2023-05-23:
**UPDATE**

I have disabled the Support for external, SPI-connected RAM in menuconfig:
[image]

and managed to start the program without any resets:
```
[1B][0;32mI (301) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [1B][0m
[1B][0;32mI (311) example: test2[1B][0m
[1B][0;32mI (311) gpio: GPIO[9]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [1B][0m
[1B][0;32mI (321) example: test3[1B][0m
[1B][0;32mI (321) example: Turn off LCD backlight[1B][0m
[1B][0;32mI (331) gpio: GPIO[38]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [1B][0m
[1B][0;32mI (341) example: test4[1B][0m
[1B][0;32mI (341) example: Initialize Intel 8080 bus[1B][0m
[1B][0;32mI (341) example: Install LCD driver of st7789[1B][0m
[1B][0;32mI (351) gpio: GPIO[5]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [1B][0m
[1B][0;32mI (481) example: Turn on LCD backlight[1B][0m
[1B][0;32mI (481) example: Initialize UI[1B][0m
[1B][0;32mI (481) example: buf1@0x3fc94c60, buf2@0x3fca4664[1B][0m
[1B][0;32mI (1481) example: trans_done_calls: 101[1B][0m
[1B][0;32mI (2481) example: trans_done_calls: 100[1B][0m
[1B][0;32mI (3481) example: trans_done_calls: 100[1B][0m
[1B][0;32mI (4481) example: trans_done_calls: 100[1B][0m
[1B][0;32mI (5481) example: trans_done_calls: 100[1B][0m
```

However, my display looks like:

[image]

KamranAghlami, 2023-05-23:
Hi @krupis.

If you're getting a 32x32 square flashing black and white it's working as intended.

This line is drawing a 32x32 square at position (32, 32) and buf is switching between buf1 (black) and buf2 (white), rest of screen isn't being drawn so it contains random data.
``` C
esp_lcd_panel_draw_bitmap(panel_handle, 32, 32, 64, 64, buff);
```

krupis, 2023-05-23:
> Hi @krupis.
> 
> If you're getting a 32x32 square flashing black and white it's working as intended.
> 
> This line is drawing a 32x32 square at position (32, 32) and buff is switching between buf1 (black) and buf2 (white), rest of screen isn't being drawn so it contains random data.
> 
> ```c
> esp_lcd_panel_draw_bitmap(panel_handle, 32, 32, 64, 64, buff);
> ```

Oh. I didint realize that! Thanks a lot for clarifying.

I have a lot of difficulty understanding how to use display functions to draw anything useful. 
For example, if I want to display text on the top left  "Random number:" and then random number generated for example: "Random number:15"

Do you plan to use LVGL?

KamranAghlami, 2023-05-23:
Sure, no problem.

I'd like to try that but as of now I'm not sure when that would be I'm busy at the moment.
If using older version of IDF with Arduino and PlatformIO is an option for you, I have an [example] (https://github.com/KamranAghlami/T-Display-S3.git) here which utilizes lvgl that you might want to try.

krupis, 2023-05-23:
> Sure, no problem.
> 
> I'd like to try that but as of now I'm not sure when that would be, I'm quite busy at the moment.
> 
> If using older version of IDF with Arduino and PlatformIO is an option for you, I have an example [here](https://github.com/KamranAghlami/T-Display-S3.git) which utilizes lvgl that you might want to try.

Thanks, il have a look.

One more thing - did you look into how to get touch display to work on esp-idf?

According to lilygo documentation, they use:
[image]

But the esp-idf only supports:
[image]

KamranAghlami, 2023-05-24:
> Thanks, il have a look.
>
> One more thing - did you look into how to get touch display to work on esp-idf?
>
> According to lilygo documentation, they use: [image]
>
> But the esp-idf only supports: [image]

The version I've got is the non-touch version, I'm not sure, If IDF doesn't natively support that chip you have to roll your own driver for it. I'd check if any of those chips that IDF supports are similar to yours, maybe you'll get lucky.

zef, 2024-05-29:
It took me some work to get this working with the latest versions of `espidf` and `lvgl`. This thread helped me figure some things out.

I packaged everything up as clean an minimally as possible and published it to a project repo. Hope it helps somebody out. There are a couple things I'd like to figure out and improve and I'll keep it up to date as I'm able while it's relevant to my work. Thanks!

https://github.com/zef/t-display-s3-espidf-lvgl