Xinyuan-LilyGO/T-Display-S3 #104: Local library copies

svdrummer, 2023-02-16:
Does anyone know how to stope local libraries within the sketch folder. Swapping libraries between S# models and other devices that use the TFT library has become painful. Even renaming the library would help.
SPI vs 8bit parallel and having two special libraries with the same name doesn't work for most people.

mmMicky, 2023-02-16:
You can update the latest version of the TFT library, which has added the driver for this board.

svdrummer, 2023-02-16:
I saw the new version..however... Doesn't lvgl and a few others have to be changed? According to the other T-displayS3 GitHub main module support site. The one with all the drawings and how to set up instructions. Maybe I am wrong.

owiecc, 2023-02-18:
The best you can do is to use PlatformIO. Then you can keep the pin and library config and inside the `platformio.ini` config file. No library swapping or modification is necessary. I will submit a PR with a minimal example soon.

closed 2023-02-22 by mmMicky