Xinyuan-LilyGO/T-Display-S3 #107: dont upgrade arduino esp32 board as yet

svdrummer, 2023-02-23:
I just upgraded to the latest esp32 on arduino.
There was an issue listed already, I just reverted back to the old version to fix.

I know it isn't a library issue. I just thought it would help users of this library and think it is a library problem.

svdrummer, 2023-02-24:
@me-no-dev The posts with this issue i have seen where always Win10/11.
Never had or have on Mac. Tried with my old Win10 (Pro x64) Laptop and had too.
Seems it is an ESPTOOL issue on windows OS as Apple/linux was not updated.
I did notice there was a new board TTGO TDisplayS3. default setting 16Meg/Huge app etc are default config with no change options.

doh38, 2023-02-26:
Yeah, I updated beginning of the week, and it broke everything. Actually in a pretty bad way since it went in bootloop... In the bootloader mode 🙈. Got lucky with an old Arduino version and the normal esp32s3-dev settings and upload an empty sketch (as soon as the old 1.8 Arduino didn't seems to see that the device was disconnecting every seconds form the USB, the 2.0.3 was unusable for this reason.
So it should be very important Imho (at least for Linux user) to be aware that the 2.0.7 is pretty dangerous right now :)

369Martin369, 2023-05-20:
Esp32 2.0.9 and IDE 2.1.0 are fine....

closed 2024-05-24 by github-actions[bot]