Xinyuan-LilyGO/T-Display-S3 #109: SD card issues with official adapter

fOmey, 2023-02-25:
I can't seem to get my SD card mounted, formatted as FAT32.. 30GB, using the officially branded [LilyGo T-display TF shield](https://www.lilygo.cc/products/t-display-tf-shied).

Factory example reports "SD card not found".

SD example:

```
E (1083) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (1083) sdmmc_sd: sdmmc_check_scr: send_scr returned 0x107
E (1084) vfs_fat_sdmmc: sdmmc_card_init failed (0x107).
Card Mount Failed
```

snowcarver75, 2023-02-28:
Been having the same issue.  Tested with an arduino and the reader works fine.   Something is off with the official code example.  Don’t see CS/SS pin being configured anywhere as well.

fOmey, 2023-02-28:
> Been having the same issue. Tested with an arduino and the reader works fine. Something is off with the official code example. Don’t see CS/SS pin being configured anywhere as well.

I'm thinking its a upstream issue, seems to be a few esp32-s3's having issues with sdcard in search results.

fOmey, 2023-03-03:
> Been having the same issue. Tested with an arduino and the reader works fine. Something is off with the official code example. Don’t see CS/SS pin being configured anywhere as well.

Managed to get a working example, can you confirm?

```
#include "FS.h"
#include "SD_MMC.h"

#define SD_CS_PIN        10 // SS / Default definition

// These pins will be use for SPI2
#define SDMMC_CLK     11
#define SDMMC_D0      12 // MOSI
#define SDMMC_CMD     13 // MISO

void listDir(fs::FS &fs, const char * dirname, uint8_t levels){
    Serial.printf("Listing directory: %s\n", dirname);

    File root = fs.open(dirname);
    if(!root){
        Serial.println("Failed to open directory");
        return;
    }
    if(!root.isDirectory()){
        Serial.println("Not a directory");
        return;
    }

    File file = root.openNextFile();
    while(file){
        if(file.isDirectory()){
            Serial.print("  DIR : ");
            Serial.println(file.name());
            if(levels){
                listDir(fs, file.path(), levels -1);
            }
        } else {
            Serial.print("  FILE: ");
            Serial.print(file.name());
            Serial.print("  SIZE: ");
            Serial.println(file.size());
        }
        file = root.openNextFile();
    }
}

void createDir(fs::FS &fs, const char * path){
    Serial.printf("Creating Dir: %s\n", path);
    if(fs.mkdir(path)){
        Serial.println("Dir created");
    } else {
        Serial.println("mkdir failed");
    }
}

void removeDir(fs::FS &fs, const char * path){
    Serial.printf("Removing Dir: %s\n", path);
    if(fs.rmdir(path)){
        Serial.println("Dir removed");
    } else {
        Serial.println("rmdir failed");
    }
}

void readFile(fs::FS &fs, const char * path){
    Serial.printf("Reading file: %s\n", path);

    File file = fs.open(path);
    if(!file){
        Serial.println("Failed to open file for reading");
        return;
    }

    Serial.print("Read from file: ");
    while(file.available()){
        Serial.write(file.read());
    }
}

void writeFile(fs::FS &fs, const char * path, const char * message){
    Serial.printf("Writing file: %s\n", path);

    File file = fs.open(path, FILE_WRITE);
    if(!file){
        Serial.println("Failed to open file for writing");
        return;
    }
    if(file.print(message)){
        Serial.println("File written");
    } else {
        Serial.println("Write failed");
    }
}

void appendFile(fs::FS &fs, const char * path, const char * message){
    Serial.printf("Appending to file: %s\n", path);

    File file = fs.open(path, FILE_APPEND);
    if(!file){
        Serial.println("Failed to open file for appending");
        return;
    }
    if(file.print(message)){
        Serial.println("Message appended");
    } else {
        Serial.println("Append failed");
    }
}

void renameFile(fs::FS &fs, const char * path1, const char * path2){
    Serial.printf("Renaming file %s to %s\n", path1, path2);
    if (fs.rename(path1, path2)) {
        Serial.println("File renamed");
    } else {
        Serial.println("Rename failed");
    }
}

void deleteFile(fs::FS &fs, const char * path){
    Serial.printf("Deleting file: %s\n", path);
    if(fs.remove(path)){
        Serial.println("File deleted");
    } else {
        Serial.println("Delete failed");
    }
}

void testFileIO(fs::FS &fs, const char * path){
    File file = fs.open(path);
    static uint8_t buf[512];
    size_t len = 0;
    uint32_t start = millis();
    uint32_t end = start;
    if(file){
        len = file.size();
        size_t flen = len;
        start = millis();
        while(len){
            size_t toRead = len;
            if(toRead > 512){
                toRead = 512;
            }
            file.read(buf, toRead);
            len -= toRead;
        }
        end = millis() - start;
        Serial.printf("%u bytes read for %u ms\n", flen, end);
        file.close();
    } else {
        Serial.println("Failed to open file for reading");
    }

    file = fs.open(path, FILE_WRITE);
    if(!file){
        Serial.println("Failed to open file for writing");
        return;
    }

    size_t i;
    start = millis();
    for(i=0; i<2048; i++){
        file.write(buf, 512);
    }
    end = millis() - start;
    Serial.printf("%u bytes written for %u ms\n", 2048 * 512, end);
    file.close();
}

void setup(){
    Serial.begin(115200);
    delay(2000);
    SD_MMC.setPins(SDMMC_CLK, SDMMC_CMD, SDMMC_D0);
    if(!SD_MMC.begin("/sdcard", true, false, 20000, 5)){
        Serial.println("Card Mount Failed");
        return;
    }
    uint8_t cardType = SD_MMC.cardType();

    if(cardType == CARD_NONE){
        Serial.println("No SD_MMC card attached");
        return;
    }

    Serial.print("SD_MMC Card Type: ");
    if(cardType == CARD_MMC){
        Serial.println("MMC");
    } else if(cardType == CARD_SD){
        Serial.println("SDSC");
    } else if(cardType == CARD_SDHC){
        Serial.println("SDHC");
    } else {
        Serial.println("UNKNOWN");
    }

    uint64_t cardSize = SD_MMC.cardSize() / (1024 * 1024);
    Serial.printf("SD_MMC Card Size: %lluMB\n", cardSize);

    listDir(SD_MMC, "/", 0);
    createDir(SD_MMC, "/mydir");
    listDir(SD_MMC, "/", 0);
    removeDir(SD_MMC, "/mydir");
    listDir(SD_MMC, "/", 2);
    writeFile(SD_MMC, "/hello.txt", "Hello ");
    appendFile(SD_MMC, "/hello.txt", "World!\n");
    readFile(SD_MMC, "/hello.txt");
    deleteFile(SD_MMC, "/foo.txt");
    renameFile(SD_MMC, "/hello.txt", "/foo.txt");
    readFile(SD_MMC, "/foo.txt");
    testFileIO(SD_MMC, "/test.txt");
    Serial.printf("Total space: %lluMB\n", SD_MMC.totalBytes() / (1024 * 1024));
    Serial.printf("Used space: %lluMB\n", SD_MMC.usedBytes() / (1024 * 1024));
}

void loop(){

}
```

snowcarver75, 2023-03-03:
Wow. Thank you that does work!

lewisxhe, 2023-03-13:
LilyGo T-display TF shield uses 1-bit SD mode, so it is not necessary to set CS pin.
You only need to set the 1-bit mode when initializing SD

closed 2023-03-13 by lewisxhe

summerfind, 2024-06-08:
> Managed to get a working example, can you confirm?
Hi, is this for T-DISPLAY or T-DISPLAY S3? my testing on the latter doesn't work, how to fix? 
nothing output.

summerfind, 2024-06-09:
Hi, is this for T-DISPLAY or T-DISPLAY S3? my testing on the latter doesn't work, how to fix?
nothing output.

teastainGit, 2024-06-09:
[image]

epiHATR, 2025-02-24:
12 is SCK pin of SPI, but why I see it's MOSI in your above code @fOmey  ?