Xinyuan-LilyGO/T-Display-S3 #111: How to reset to factory settings after Burn Bootloader?

headkit, 2023-03-01:
Is there a way to reset the board to factory settings after Burn Bootloader?
Thnx!

teastainGit, 2023-03-01:
Hi!
I am curious what you mean by after Burn Bootloader ?
The board comes from LilyGO with a boot loader already installed and I believe it is permanently installed in ROM.
What happened to yours?
And what do you mean by factory settings?

I would like to help but I would like some more info.

How are you programming it? Arduino IDE, PlatformIO, ESP-IDF, Python?

Cheers !
Terry Stainton

***@***.***

headkit, 2023-03-02:
Thank you for your help.
I had problems to upload Code via Aruino IDE app on macOs 12.6 Monterey. I got the error

```
bash: /esptool: No such file or directory
exit status 127

```
Testing things and clicking to fast I accidently fired 'Burn Bootloader'. I think I flashed it. Now I am stuck at all, trying to reconnect to the board, loosing the port one second afte selecting it.
I don't get it to work again: When connecting Port, Serial Monitor says 
`invalid header: 0xffffffff`
then connection is gone again.

headkit, 2023-03-02:
WIth esptool I am able to connect and 
`python3 -m esptool read_flash 0 0x400000 tmp.bin`
downloads an image. (found this [here](https://samg.net.au/tutorials/installing-esptool-on-macos-and-programming-an-esp8266-witty/) - don't know why to use 0x400000)

I think I could use write_flash to fix this. Is there a basic bin file that would reset the flash to factory settings?

headkit, 2023-03-02:
I also found this option: [CONFIG_BOOTLOADER_FACTORY_RESET](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/kconfig.html#config-bootloader-factory-reset) but don't know how to use it.

doh38, 2023-03-03:
@headkit check which library you are using. I had a very similar (bad situation) while updating this week to the last esp-32 direver (2.0.7) I reverted back to 2.0.6, put it in bootloader mode, used old 1.8.19 arduino ide (which keeps the port connection somehow even if the esp is disconnecting from usb)  and uploaded from there an empty neutral sketch (only setup and loop empty). this saved it. 
Hope that helps :)

headkit, 2023-03-03:
I guess I need to find out first what bootloader mode means. I am such a beginner to board dev. Arduino IDE is acting weird, losing connection every second.
In most cases its a wrong path to sources, but that should not be the problem. I dono't get it why Apple changes everything every year so that finding matching resources is such a pain.

I think they changed the IDE to use VS Code in the background. I will try to use an older IDE.
thnx

headkit, 2023-03-03:
> @headkit check which library you are using. I had a very similar (bad situation) while updating this week to the last esp-32 direver (2.0.7) I reverted back to 2.0.6, put it in bootloader mode, used [...]

using Arduino 1.8.19 I get the error

`Caused by: java.io.IOException: error=2, No such file or directory`

when uploading example sketches, like 'Sprite_draw'.

headkit, 2023-03-05:
I deinstalled v2.0.6 and reinstalled 2.0.7 of **esp32 by Espressif Systems**. Using Arduino IDE 2.0.4 I still lose USB connection right after connecting port.

headkit, 2023-03-06:
@teastainGit I now can upload and run code using Arduino 1.8.19 and 2.0.7 of esp32 by Espressif Systems. 
Arduino IDE still is losing port connection right after selection.

headkit, 2023-03-06:
I now also put the device into "Upload mode" and it seems that Arduino IDE is establishing a stable connection again.

closed 2023-03-07 by headkit

e-p-s, 2023-06-16:
what is the upload mode?

i flashed mine and now i have a boot loop. the usb pling pling from windows repeats and repeats.
and the red flash led is blinking all the time

lewisxhe, 2023-06-16:
@e-p-s  https://github.com/Xinyuan-LilyGO/T-Display-S3#quick-start

e-p-s, 2023-06-16:
can't see the wood for the trees

LucasEaston, 2023-06-23:
> can't see the wood for the trees
This helped me fix mine. The BOOT button is the bottom Right button when holding the unit horizontally.
I'm assuming this puts it in a mode where it doesn't run whatever broken sketch you have accidentally uploaded haha. So it boots in an idle state, so you can replace the broken sketch without the unit crashing constantly. Then when you hit reset it will run the fresh sketch.

1.Connect the board via the USB cable
2.Press and hold the BOOT button , While still pressing the BOOT button, press RST
3.Release the RST
4.Release the BOOT button
5.Upload sketch
6.Once sketch is uploaded, press and release RST

1liminal1, 2025-03-18:
Hey guys, I'm trying to install a WLED bin. I used esptool to install; 

python -m esptool --chip esp32s3 --port COM7 --baud 460800 write_flash 0x0 "C:\wled_backup\WLEDMM_14.5.0-beta_esp32S3_16MB_PSRAM_M_HUB75.bin"

But its just in a bootloop, do I need a bootloader, how can I get this installed correctly

lewisxhe, 2025-03-18:
If a loop occurs, then there is basically a problem with the firmware. You can try to write another firmware to test it.

1liminal1, 2025-03-18:
So im trying to load a WLED bin file on it. 

If I write an arduino sketch to flash the LED, its fine. But when I try to load a bin file. Not fine.

1liminal1, 2025-03-18:
Its ok, thank you :) 

https://wled-install.github.io/

I got it working using these webtools