Xinyuan-LilyGO/T-Display-S3 #113: Color calibration Issue (reds are Orange)

doh38, 2023-03-03:
Hi,
The colors are quite off which is a bit sad seeing the good quality of the display.
-To make it short. it's impossible to get an acceptable "red". 0xF800 is Orange at the best.
-Light_Grey 0xCDF9 for instance is quite ok when looking close to the display, but will emit green tint when  the "observant" goes a bit away from screen.
I have seen some kind of gamma settings/ color correction tool in the ST7789V driver. But I have no Idea how to correct this. Is there some "tutorial/info?" 
Or it is an Hardware thing I may have to live with it?
Gruß,
[image]

teastainGit, 2023-03-03:
Me too!
[image]

lewisxhe, 2023-03-13:
Sorry, the current initial gamma value is provided by the screen manufacturer. I don't know how to calibrate the gamma value. These values are generally provided by the screen manufacturer. I have already fed back to the screen manufacturer to see if they have any solutions

doh38, 2023-03-18:
very good, please et us know when you get a reply. it's actually very present, and as soon as LilyGo sells a "display" it's somehow mandatory that it somehow shows color close enough to expectations. As it is now, the reds are simply impossible to draw except with some "hacking" of rgb values to compensate (only at some brightness settings) the mismatch. :)

closed 2024-05-22 by github-actions[bot]