Xinyuan-LilyGO/T-Display-S3 #116: BootLoader questions

teastainGit, 2023-03-07:
Many people (including myself) who start working with the LilyGO T-Display S3 have difficulty uploading or experience no image on LCD, no serial data on serial monitor, etc. They feel upset that they have "bricked" their new board.
A small number of people feel that they have erased or corrupted the BootLoader and then go to GREAT lengths trying to erase and restore to "Factory Settings". They attack the board aggressively  with various burner programs and utilities from Espressif and who-knows-what from The Internet.
So, my question is:
Is "The BootLoader" permanently stored in some safe ROM section of the ESP32?
Under what circumstances could it become Bricked?
If you are using Arduino V2.0.4 and ESP32 v2.0.6 which some consider "lame" is there a scenario where:
entering upload mode with the Boot  and Reset buttons,
and uploading a simple known test program,
this will not work?
Thanks, Terry

reflectingme, 2023-03-12:
@teastainGit - Hey Terry - are you by any chance using TFT_eSPI libraries in your code?  if so, it might be due to this library.  If you are using TFT_eSPI then you need the specific library for the T-Display board you are using and NOT the general TFT_eSPI version.

Not sure that is what is causing your issue, but I thought I'd drop a reply as a just in case.  

John

lewisxhe, 2023-03-13:
The boot program of esp32 is in the OPT area of the chip. It can only be programmed once and can never be changed. Please rest assured that it will not become a brick
When it cannot be written or runs abnormally, you need to press and hold the BOOT key, and then insert the USB cable. At this time, you can basically solve the problem by clicking Upload again
The above is the way to enter the download mode

closed 2023-03-13 by lewisxhe

Stark0008, 2023-11-08:
Has anyone found a solution to the problem? When updating the firmware with Arduino, the display stopped working, I tried different libraries, I also tried different versions of the esp32 board, including 2.0.6. When flashing .bin files via flash_download_tool_3.9.5 everything works fine. What could be the problem?

Stark0008, 2023-11-08:
> Многие люди (включая меня), которые начинают работать с LilyGO T-Display S3, испытывают трудности с загрузкой или не видят изображения на ЖК-дисплее, последовательных данных на последовательном мониторе и т. д. Они расстраиваются из-за того, что «закирпичили» свою новую плату. Небольшое количество людей считают, что они стерли или повредили загрузчик, а затем прикладывают все усилия, пытаясь стереть и восстановить «заводские настройки». Они агрессивно атакуют плату с помощью различных программ записи и утилит от Espressif и черт знает чего из Интернета. Итак, мой вопрос: «The BootLoader» постоянно хранится в каком-то безопасном разделе ПЗУ ESP32? При каких обстоятельствах он мог стать кирпичом? Если вы используете Arduino V2.0.4 и ESP32 v2.0.6, которые теперь стабильны и надежны, существует ли сценарий, при котором: вход в режим загрузки кнопками Boot и Reset, и загрузив простую известную тестовую программу, что плата не будет работать? Спасибо, Терри

Has anyone found a solution to the problem? When updating the firmware with Arduino, the display stopped working, I tried different libraries, I also tried different versions of the esp32 board, including 2.0.6. When flashing .bin files via flash_download_tool_3.9.5 everything works fine. What could be the problem?