Xinyuan-LilyGO/T-Display-S3 #117: No way to get a DS18B20 sensor working when connected on the SH1.0 connector (GPIO43)

Rollmops67, 2023-03-08:
I'm rather familiar with I2C sensors like the AHT20 (using the "Wire.h" library), and also with oneWire sensors like the DS18B20 (using the "oneWire.h" library). I have several projects wich I use one or the other sensor type, and had never a problem.
On the T-Display I allready did projects where I connect an I2C sensor to the 4pin SH1.0 connector just under the boot button.
The `Wire.begin(43, 44);` command in the setup does the job, and it works just fine.

Today I wanted to connect a DS18B20 oneWire sensor on this connector (usin GPIO43 pin).
Of course I declared the pin in the sketch with the `#define oneWireBus 43` command.
I also soldered a 5 kohm pullup resistor between GPIO43 and +3.3 V.
But it just didn't work, the DS18B20 sensor returns a value of -127°C, meaning it is not read.
I tried 3 different DS18B20 sensors (2 different types), but nothing to do, it didn't work too.

Just to be sure I tried it with GPIO21 (of course with changing the `#define oneWireBus 43` to `#define oneWireBus 21`) , and voilà, now it's OK, the sensor works as advertised.

I have no idea why it works on GPIO21 and not on GPIO43.
Did somebody allready succeed in connecting a DS18B20 on GPIO43 ?

Roland

teastainGit, 2023-03-08:
Hi!
I assume you also soldered a pull-up when you tried GPIO21?
ESP32 has built in pullups, maybe try higher value (weaker) resistor?
Barring that it is strange but the LilyGO pinup shows 43 as a i2c pin and 21 isn't?
Both are 'generic' GPIO so this should be no problem.
"That's all I got!"

teastainGit, 2023-03-08:
OK< I'm back. The pins 43 and 44 are dedicated RX and Tx for UART or i2c.
Neither will work with my DS18B20 for me, but...
My sensor works fine on GPIO16, (which I already had a connection) to test.
Disappointing result for you, but good to know for future!
Cheers, Terry

Rollmops67, 2023-03-09:
Hello teastainGit, thank you for your answers !

Concerning your first answer :
Of course I soldered the pullup on pin 21, I also tried with only 2,5 Kohms...
It was 3-wires mode.
I allready did several projects with DS18B20 sensors, so no problem with the software nor with the hardware here !

Concerning your second answer :
OK I'm reassured now !
My I2C sensors also work fine on pins 43 and 44.
So I simply replaced the DS18B20 with an I2C based AHT20 sensor, no problem with this one.

Thank you for the time spent !

Roland

closed 2023-03-09 by Rollmops67

cc13com, 2024-02-10:
Hello,

can I jump in here please? I'm also trying to get a DS18B20 running on my LilyGo. For that I got a ready configured sensor with a small pluggable terminal with small electronic on it. On another board I can connect VCC, Ground and Data to the pins directly. 

Unfortunately I run in the following error on the serial monitor and the board is rebooting always:
```
[   224][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
E (50) gpio: gpio_set_level(226): GPIO output gpio_num error
[   225][E][esp32-hal-gpio.c:102] __pinMode(): Invalid pin selected
E (56) gpio: gpio_set_level(226): GPIO output gpio_num error
```

Here a part of my code:
```
...
#include <OneWire.h>
#include <DallasTemperature.h>
...
#define SENSOR_PIN  16
#define DEBUG true
OneWire oneWire(SENSOR_PIN);
DallasTemperature DS18B20(&oneWire);

void setup() {
  Serial.begin(460800);
  DS18B20.begin(); 
...
deviceCount = DS18B20.getDeviceCount();
```

I don't have a SH1.0-Plug on my temperature sensor therefore I will connect each single cable (VCC, Ground) with the board and the Data cable with GPIO 16. I also tried GPIO 17 and GPIO 2 with the same result. 

But if I print the "deviceCount" on the console it shows me that 1 sensor was found.

Any idea what's wrong from my side?

Carsten

teastainGit, 2024-02-10:
Here is my LilyGO T-Display S3 running a DS18B20 on pin16!
```
#include <DallasTemperature.h>
OneWire oneWire(16);  //the pin you want to use
DallasTemperature sensors(&oneWire);

void setup() {

#include <OneWire.h>
  Serial.begin(115200);
  delay(200);
  Serial.println(" setup ");
  delay(200);
  sensors.begin();
}

void loop() {
  sensors.requestTemperatures();
  float temperatureC = sensors.getTempCByIndex(0);
  float fahrenheit = (temperatureC * 9.0 / 5.0) + 32.0;
  Serial.print(temperatureC);
  Serial.println("ºC");
  Serial.println("   ");
  delay(2000);
}
```

[image]

 Try this and compare with your code!
Arduino IDE 2.2.1, all libraries and board definitions up-to-date.
-Terry

cc13com, 2024-02-11:
Hello @teastainGit,

thanks for your example code. It's working fine. From this I started to implement my needed features (WIFI, MQTT, TFT and more then one DS18B20 device) step-by-step.

Wifi works. MQTT works. But as soon as I include `#include <TFT_eSPI.h>` with the needed code to init and show something on the display the board is resetting again.

If I compile a small example just to display one line on the display without any other library it works too. It's "TFT_eSPI Version 2.5.34".

cc13com, 2024-02-11:
One additional point: If I run the sketch with the full code (include TFT, OneWire, WiFi, PubSubClient, DallasTemperature) but without a DS18B20 connected, the board will not reset. As soon as I connect the 3 pins (VCC, GND, DAT) from the temp sensor to the board it's resetting every two seconds.

teastainGit, 2024-02-11:
OK, I added the sensor to HotHead with pin 16.
[HotHead_DS18B20_Pin16.zip](https://github.com/Xinyuan-LilyGO/T-Display-S3/files/14233614/HotHead_DS18B20_Pin16.zip)

See how it runs!
What 'User_Setup_Select.h' are you using in TFT_eSPI  library?
I included the correct one in the .zip

cc13com, 2024-02-17:
I could start your example and I'm sure that I did the change in 'User_Setup_Select.h' in my own sketch before. Because the TFT was working fine but not together with the temperature sensor. In your example it's just the TFT and the sensor, but no WiFi and no MQTT.

That's what I will add now.