Xinyuan-LilyGO/T-Display-S3 #118: Intermittent color glitch on T-Display-S3

tonystuart, 2023-03-10:
Just wanted to start by saying I think the T-Display-S3 is a great board at a great price point. I don't think the problem I'm describing here is necessarily related to the T-Display-S3, only that it's showing up on a project I built using the T-Display-S3.

I designed a circuit board for the T-Display-S3 and had it produced by JLCPCB. Everything is working great except I see an intermittent display glitch. I'm using my own graphics code and a driver based on the Espressif esp-idf i80 bus and ST7789 driver APIs.

The glitch shows up as a short (generally about a second or less) switch to alternate colors and back. For example if the expected colors are yellow on red, it switches briefly to purple on green, and then back.

The text and graphics on the display are not changed, only their color.

I see there are lot of questions on the web about configuring a system to display colors correctly on the ST7789. This is a different problem, in that the colors are correct except for this intermittent glitch.
 
During initialization, I invoke `esp_lcd_panel_invert_color(handle, true)` to set the expected colors. After that, I don't invoke it again. I see that function uses the display HAL to send the following command to the ST7789 using `esp_lcd_panel_io_tx_param`:

`#define LCD_CMD_INVON        0x21 // Go into display inversion mode`

It's almost as if the reverse of this command is getting processed intermittently.

It doesn't happen very often and I'm having trouble correlating it with anything that the board is doing, but some factors that I think may cause it include:

- The amount of time the board has been on (possibly related to temperature)
- The combined processor workload (possibly related to bus clock, interrupts, etc.)
- Other things happening at the same time on the circuit board (possibly related to power supply)

I'm thinking this is probably due to my specific hardware and firmware environment and I'll probably have to figure it out myself, especially without providing a lot more detail and/or a reproducible test case.

But before I get too far down that path I wanted to check to see if anyone else is ever seeing anything like this, or if someone with more experience with the ST7789 and display can offer any suggestions about what might cause the colors to change.

Thanks!

lewisxhe, 2023-03-13:
I have not encountered this situation. I have added a tag to your issue. I hope someone can help you

ruycesar, 2023-04-15:
I'm a seasoned EE with decades of circuit design and debug experience but I still find myself using the "Finger test method" to quickly identify the source of this kind of issues (or eliminate some probable sources). It consists in running one or two fingers through you live PCB, touching exposed component pads and, at the same time, observing the display to see if the issue manifests itself with more frequency or less frequency depending on where you touch. Of course, you always have to make sure you discharge any static electricity you may have in your hands BEFORE touching any components. And Also, don't try touching anything which may have voltages higher than 12 Volts. Just common sense. The Human Finger electrical model is very complex and it could act as a HF filter/damper, a pull up or pull down, a current return path for high speed signals, etc. If the behavior does not change during the finger test, most likely the problem is in the software. But if it does change, you have a hardware problem in your hands (no pun intended)

closed 2024-05-22 by github-actions[bot]