Xinyuan-LilyGO/T-Display-S3 #120: Pinout diagram for the LilyGo T-Display TOIUCH version please

reflectingme, 2023-03-12:
So sorry folks, I have seen this diagram but for the life of me I can't seem to find it any more.  I am sure I have seen a diagram on LilyGo GitHub [page somewhere that shows mainly the T-Display, but there is an added image on that page showing the additional Touch screen pinouts.  Could someone please put me out of my misery and point me to that page.

Ta.

szucsz, 2023-03-12:
do you mean this? https://github.com/Xinyuan-LilyGO/T-Display-S3/blob/main/schematic/T_Display_S3.pdf

Or something else?

reflectingme, 2023-03-12:
> do you mean this? https://github.com/Xinyuan-LilyGO/T-Display-S3/blob/main/schematic/T_Display_S3.pdf
> 
> Or something else?

Alas, something else @szucsz sorry.  I have seen a diagram of the pinout somewhere on their pages, but I just can't seem to locate it now!  Thanks for this though.

reflectingme, 2023-03-12:
It was something like this one

[image]

But with an added diagram / image of the Touch version showing the pinouts.

teastainGit, 2023-03-12:
https://github.com/Xinyuan-LilyGO/T-Display-S3 
The ReadMe shows a pinout including the touch inputs.
Is this what you are talking about? GPIO 1-3 and 10-13.

reflectingme, 2023-03-12:
No Terry.  I am certain I saw an image, which was the one in the readme file, but in the lower right hand side of the image there was an additional picture of the physical TOUCH version, with the pinout shown on that.

lewisxhe, 2023-03-13:
Is that what you mean?
https://github.com/Xinyuan-LilyGO/T-Display-S3/blob/main/image/T-DISPLAY-S3-TOUCH.png

reflectingme, 2023-03-13:
> Is that what you mean?
> https://github.com/Xinyuan-LilyGO/T-Display-S3/blob/main/image/T-DISPLAY-S3-TOUCH.png

Yes, Lewis, that's the one.  Thank you very much.

And, I am certain that this was not on the GitHub page for LilyGo T-Display yesterday!  However, it is the image I was referring to, thanks.

closed 2023-03-13 by reflectingme