Xinyuan-LilyGO/T-Display-S3 #121: Using the original 2.5.23 TFT_eSPI

daniel-prause, 2023-03-15:
Hi there!

Is it possible to use the official TFT_eSPI library?
From my testing yesterday, unfortunately it does not seem to be the case.
When I use the TFT_eSPI lib from here, everything works fine as expected:

[image]

When I use the official one, this happens:

[image]

What things were changed in the TFT_eSPI library provided in this repository, that had an impact on the offset?
When I tried to figured it out yesterday, there wasn't anything giving me clue which could be the problem with the official library.

From what I could find out, it is based on the 2.5.0 version of TFT_eSPI from Bodmer.

Thanks in advance for providing me some information! 👍 

Daniel

teastainGit, 2023-03-15:
Not sure I understand the problem, but let me add my thoughts:
When the board was released it had a TFT_eSPI library 2.4.71 (or so) and it was modified (forked) by LilyGO for use only on the LilyGO T-Display S3. MANY users on this Issues forum were frustrated that it was unique and of the same name, meaning that other boards could not use it and we had to constantly switch around.
LilyGO FINALLY worked with Bodmer to produce V2.5.0 which is now in the LilyGO repository. This new library was now universal.
But...Arduino has still not provided 2.5.0 as a library update and they are stuck at 2.4.79. 
So you get your board running and then do a Library Update on all your libraries and then LilyGO T-Display S3 won't run!
-Terry

pdurys, 2023-03-15:
You can use e.g. Notepad++ to compare files content and find yourself what is different with standard TFT_eSPI and that one provided here. Do you need to know how to make such comparison or LMGTFY?

daniel-prause, 2023-03-15:
> You can use e.g. Notepad++ to compare files content and find yourself what is different with standard TFT_eSPI and that one provided here. Do you need to know how to make such comparison or LMGTFY?

It could have been possible, that one probably knows, what is causing the issue - judging by the differences between the TFT_eSPI 2.5.0 provided in this repository and the latest official version 2.5.23.

The thing is - the information one gets is scattered everywhere e.g. within the PlatformIO repository, there is a TFT_eSPI library with the version 2.5.23 which has e.g. the LilyGO S3 Display as an uncommentable header within the User_Setup_Select.h.
If I use that version instead of the version provided inside this repository here, the offset you can see in the pictures that I have posted gets in effect.
That seems to me as if the "official" TFT_eSPI library probably is either not compatible with the S3 display or there is something that I don't know of needs to be configured.

daniel-prause, 2023-03-15:
Update:
For anyone who falls into the same rabbit hole as I did, the things that did the trick for me were:

1) Add these lines to your platformio.ini:

build_flags =
	-D BOARD_HAS_PSRAM
	-D ARDUINO_USB_MODE=1
	-D ARDUINO_USB_CDC_ON_BOOT=1
	;
	-D USER_SETUP_LOADED=1
;	*** Setup206 _LilyGo_T_Display_S3.h ***
	-D ST7789_DRIVER
	-D CGRAM_OFFSET
	-D TFT_RGB_ORDER=TFT_BGR
	-D TFT_INVERSION_ON
	-D TFT_PARALLEL_8_BIT
	-D TFT_WIDTH=170
	-D TFT_HEIGHT=320
	-D TFT_DC=7
	-D TFT_RST=5
	-D TFT_WR=8
	-D TFT_RD=9
	;
	-D TFT_D0=39
	-D TFT_D1=40
	-D TFT_D2=41
	-D TFT_D3=42
	-D TFT_D4=45
	-D TFT_D5=46
	-D TFT_D6=47
	-D TFT_D7=48
	;
	-D TFT_BL=38
	-D TFT_BACKLIGHT_ON=HIGH
	;
	-D LOAD_GLCD
	-D LOAD_FONT2
	-D LOAD_FONT4
	-D LOAD_FONT6
	-D LOAD_FONT7
	-D LOAD_FONT8
	-D LOAD_GFXFF
	-D SMOOTH_FONT

These were copied from https://github.com/Bodmer/TFT_eSPI/discussions/2010#discussioncomment-4620612 
Thanks to https://github.com/Bodmer/TFT_eSPI/discussions/2010#discussioncomment-4620612!!!

For me, there was no need to tinker with TFT_eSPI when using 2.5.23 and these settings in the platformio.ini.