Xinyuan-LilyGO/T-Display-S3 #122: lib/TFT_eSPI/User_Setups/Setup206_LilyGo_T_Display_S3.h needs RGB order swapped

daavery, 2023-03-16:
#define TFT_RGB_ORDER TFT_RGB  // Colour order Red-Green-Blue
//#define TFT_RGB_ORDER TFT_BGR // Colour order Blue-Green-Red

daavery, 2023-03-16:
T-Display S3 ( non touch)

lewisxhe, 2023-03-21:
There are two versions of the screen, and my display is correct. When did you buy yours?

daavery, 2023-03-21:
2 weeks ago - amazon - X003EN0D15

-- 
David A. Avery
Avery Design
  Technical Theatre
    Lighting
    Scenery Construction
    Sound Design
    Live Recording
   (303) 725-7515 (cell)

lewisxhe, 2023-03-21:
Please repurchase the warehouse again. Is it really normal? Because my display with and without touch is normal

lewisxhe, 2023-03-21:
[image]

closed 2024-05-22 by github-actions[bot]