Xinyuan-LilyGO/T-Display-S3 #123: wifi is vera weak

Arielhh, 2023-03-17:
I am looking for advice on why the wifi is so weak on this device.
I am using the internal antenna. I tried different T-Diaplay S3 units with the same results.
Other ESP32 development boards (which are not T-Display S3) are working fine.
I also tried an external antenna (moved the 0 ohm resistor to the right location) with the same
poor results.

lewisxhe, 2023-03-21:
Can you provide a picture of the antenna section?

Arielhh, 2023-09-20:
You can see it here: https://www.lilygo.cc/products/t-display-s3

Arielhh, 2023-09-20:
> Can you provide a picture of the antenna section?

You can see it here: https://www.lilygo.cc/products/t-display-s3

kdg3737, 2023-12-29:
This is terrible, also seeing ping times of 200ms and more. Tested with bluetooth disabled, on both ESPIDF and Arduino, same result. Looks like a hardware issue. Please Lilygo check your Wifi signal before throwing these boards in production. It's a great board but this weak Wifi signal makes it unusable for +95% of applications.

richardsg307, 2024-01-04:
I've spent several days replacing my old ESP8266 Boiler monitor with a brand spanking new T-Display-S3, Tasmotised and showing lots of useful data, only to install it and find it won't connect to my WiFi! I've ordered an Antenna to see if it makes a difference; judging from other comments it won't. Anyway, do I need to remove a jumper or something?

Arielhh, 2024-01-05:
Agree, wifi is very bad.  in ESP32 S2 there was no Bluetooth support and it was working perfectly. Maybe there is a collision between Bluetooth and wifi. Is it possible to turn off completely the Bluetooth?

lauba11, 2024-01-08:
I have had the same experience with this board. Has anyone improved the situation with the WiFi signal?:)

richardsg307, 2024-01-08:
I bought a Tapo WiFi Extender of Amazon for £14.99 (2.4 Band only). It show 1 bar of 5 for the incoming wifi strength in my boiler location but the T-Display-S3 connects to it solidly.

closed 2024-05-20 by github-actions[bot]

besanur, 2024-07-01:
i have the same problem, very weak wifi, i have tested my custom code also some other firmware such nerdminer.. my wemos s2-mini has much better wifi quality

Arielhh, 2024-07-01:
> i have the same problem, very weak wifi, i have tested my custom code also some other firmware such nerdminer.. my wemos s2-mini has much better wifi quality

Use external antenna

besanur, 2024-07-02:
no i don't want to use external antenne, i will use another device and send this back

kdg3737, 2024-07-03:
I noticed ALL my S3 boards have this weak wifi issue, not just LilyGO. Sticking to S2 for now and praying they'll fix it in the S4.