Xinyuan-LilyGO/T-Display-S3 #124: Stepper.h failure to drive a stepper motor

cahudson, 2023-03-17:
When using stepper.h built in arduino library with a 5V 28BYJ-48 with a separate power source and ULN2003 driver board the t-display-s3 fails to drive the motor. If I code it directly with four pins and without the stepper.h library then I can move the stepper motor.

teastainGit, 2023-03-17:
I set up a circuit just now and it seems to work OK for me.
Let me know if you still need help.
I and using the Arduino Library, a Chinese ULN2003A board and a ~6V battery
[image]

cahudson, 2023-03-17:
That is pretty much the set up I am using hardware wise and have tried a uln2003a and a uln2003 board and different steppers with no luck. Are you able to post your ino code? It must be something silly I am doing in the arduino code. thanks

teastainGit, 2023-03-17:
I'm still not happy with the RPM and it does not reverse, so here is my prelim-in-progress code!
/* Stepper Motor Control - one revolution
 This program drives a unipolar or bipolar stepper motor.
 The motor is attached to digital pins 43, 44, 18, 17 of the LilyGO T-Display S3.
 The motor should revolve one revolution in one direction, then
 one revolution in the other direction.
 Created 11 Mar. 2007
 Modified 30 Nov. 2009
 by Tom Igoe
 Mosified for LilyGO
 by Terry Stainton
 Mar17, 2023
 */
#include <Stepper.h>
const int stepsPerRevolution = 200;  // change this to fit the number of steps per revolution
// for your motor
// initialize the stepper library on pins 8 through 11:
Stepper myStepper(stepsPerRevolution, 43, 44, 18, 17);

void setup() {
  // set the speed at 60 rpm:
  myStepper.setSpeed(60);
  // initialize the serial port:
  Serial.begin(115200);
}

void loop() {
  // step one revolution  in one direction:
  Serial.println("clockwise");
  myStepper.step(stepsPerRevolution);
  delay(1000);

  // step one revolution in the other direction:
  Serial.println("counterclockwise");
  myStepper.step(-stepsPerRevolution);
  delay(1000);
}

teastainGit, 2023-03-17:
Changed this value, "swapping pins" in code
Stepper myStepper(stepsPerRevolution, 43, 18, 44, 17);  //reverses now!
This seems quite useable now.

cahudson, 2023-03-17:
Ok, seeing weird behavior. I have mimic-ed your hardware set up with the same pins. I changed the steps to 200 (the actual steps are 4096 it looks like). When I run the code you gave it went forward and reverse (without switching pins) but only about a tenth of the way around a revolution. When I change to a step number any higher I get no movement from the motor and noises from the uln board (tried two different boards). Are you getting a full revolution with your code and changing the steps to match your motor? Next is to try changing out my power converter board as I am wondering if it has enough amps.

teastainGit, 2023-03-17:
const int stepsPerRevolution =2038; and myStepper.setSpeed(10);
cut-and-try this is giving 360 degree rotation CW and CCW, but rather leisurely! 
Beyond that we are skipping the phases ahead faster than it can turn.
Your experience may vary!

cahudson, 2023-03-17:
Will try tonight and see thanks - that makes sense

cahudson, 2023-03-18:
Thanks that did it - slowing down the speed solved the issue thanks so much.

closed 2023-03-20 by cahudson