Xinyuan-LilyGO/T-Display-S3 #125: no Touch IC found

369Martin369, 2023-03-17:
1) I copied all libs to my lib folder
2) Then I compile "....T-Display-S3-main\example\touch_test\touch_test.ino" 
3) Logo is flashing and I see 100ms a green text but then display turns black Serial and serial print says:
---------------------------------------------------
Hello T-Display-S3
226): GPIO output gpio_num error
E (514) gpio: gpio_set_level(226): GPIO output gpio_num error
Touch IC not found
---------------------------------------------------

What do I do wrong??

369Martin369, 2023-03-18:
Is my unit broken? I checked SCL=18 and SDA=17, so compiling should be fine

AlexanderDragosch, 2023-03-18:
same error here.
any new insigthts ?

AlexanderDragosch, 2023-03-18:
worked after i changed to "SELF" 
```
//#define TOUCH_MODULES_CST_MUTUAL
#define TOUCH_MODULES_CST_SELF
```

369Martin369, 2023-03-18:
thank you lex. can confirm function now. 
PS. P value is always 0000...so pressure does not work. However, axes are okay.

369Martin369, 2023-03-18:
Lilygo, please patch file "touch_test.ino" with 
//#define TOUCH_MODULES_CST_MUTUAL
#define TOUCH_MODULES_CST_SELF

Thanks

lewisxhe, 2023-03-21:
The t-display s3 has two types of touch screens, with a small hint at the top of the sketch that may not be obvious enough

lewisxhe, 2023-03-21:
I have already marked a clear hint above the sketch

closed 2023-03-21 by lewisxhe

369Martin369, 2023-03-21:
can you explain about P value is always 0000... and why multiple fingers are not detected? It this because of new chip?

lewisxhe, 2023-03-22:
Most TFT touch screens do not support multi touch

369Martin369, 2023-03-23:
Please a bit more specific....so you ship a mix ...some work,.some not, depending on chips used, or what???thanks

369Martin369, 2023-03-23:
[image]

2nd. You ship box with picture of a keyboard. It would be nice if you can share the code of the keyboard with us, so that we can speed up our developments. Thanks a lot!

369Martin369, 2023-03-23:
> Keyboard, please look here https://github.com/Xinyuan-LilyGO/T-Embed/tree/main/example/factory

I have no idea which code lines you refer to...

Ton-O, 2023-11-21:
Hi, I have a LilyGo T-display S3 and want touch to work. according to the seller, the touch chip has been changed from CST328 to CST816... What settings should I use for touch i that case?