Xinyuan-LilyGO/T-Display-S3 #129: "Invalid MICROPY_BOARD specified: LILYGO_T-DisplayS3"

mcctuxic, 2023-03-24:
I cloned the repository onto my harddisk, copied config_T-DisplayS3 to config and did a make.
After a while the compilation ended with this error message:
Added the following directories to PATH:
  /home/user/MICROPYTHON-TTGO-DISPLAY-S3/lilygo-micropython.build/build_dir/esp32s3/esp-idf/components/esptool_py/esptool
  /home/user/MICROPYTHON-TTGO-DISPLAY-S3/lilygo-micropython.build/build_dir/esp32s3/esp-idf/components/espcoredump
  /home/user/MICROPYTHON-TTGO-DISPLAY-S3/lilygo-micropython.build/build_dir/esp32s3/esp-idf/components/partition_table
  /home/user/MICROPYTHON-TTGO-DISPLAY-S3/lilygo-micropython.build/build_dir/esp32s3/esp-idf/components/app_update
  /home/user/.espressif/tools/xtensa-esp-elf-gdb/11.2_20220823/xtensa-esp-elf-gdb/bin
  /home/user/.espressif/tools/riscv32-esp-elf-gdb/11.2_20220823/riscv32-esp-elf-gdb/bin
  /home/user/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin
  /home/user/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s2-elf/bin
  /home/user/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin
  /home/user/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin
  /home/user/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin
  /home/user/.espressif/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/bin
  /home/user/.espressif/python_env/idf4.4_py3.10_env/bin
  /home/user/MICROPYTHON-TTGO-DISPLAY-S3/lilygo-micropython.build/build_dir/esp32s3/esp-idf/tools
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

make[1]: Entering directory '/home/user/MICROPYTHON-TTGO-DISPLAY-S3/lilygo-micropython.build/build_dir/esp32s3/micropython/ports/esp32'
idf.py -D MICROPY_BOARD=LILYGO_T-DisplayS3 -B /home/user/MICROPYTHON-TTGO-DISPLAY-S3/lilygo-micropython.build/build_dir/esp32s3/build-LILYGO_T-DisplayS3 -DUSER_C_MODULES=/home/user/MICROPYTHON-TTGO-DISPLAY-S3/lilygo-micropython.build/extmod/micropython.cmake build
Executing action: all (aliases: build)
Running cmake in directory /home/user/MICROPYTHON-TTGO-DISPLAY-S3/lilygo-micropython.build/build_dir/esp32s3/build-LILYGO_T-DisplayS3
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DMICROPY_BOARD=LILYGO_T-DisplayS3 -DUSER_C_MODULES=/home/user/MICROPYTHON-TTGO-DISPLAY-S3/lilygo-micropython.build/extmod/micropython.cmake -DCCACHE_ENABLE=0 /home/user/MICROPYTHON-TTGO-DISPLAY-S3/lilygo-micropython.build/build_dir/esp32s3/micropython/ports/esp32"...
CMake Error at CMakeLists.txt:18 (message):
  Invalid MICROPY_BOARD specified: LILYGO_T-DisplayS3

-- Configuring incomplete, errors occurred!
cmake failed with exit code 1
make[1]: *** [Makefile:34: all] Error 2
make[1]: Leaving directory '/home/user/MICROPYTHON-TTGO-DISPLAY-S3/lilygo-micropython.build/build_dir/esp32s3/micropython/ports/esp32'
make: *** [Makefile:36: all] Error 2
[1]    17526 exit 2     make

How can I successfully compile TTGO-Display-S3 on my GENTOO Linux?

albertas, 2023-04-02:
I get the same error for `LILYGO_T-Display` on Ubuntu 20.04 LTS: `Invalid MICROPY_BOARD specified: LILYGO_T-Display`.

codiak6335, 2023-09-05:
Latest pull from github.... all dependencies fulfilled... Ubuntu 23.04... cp t-displays3 to config, make... and  this

_-- DEBUG: Use esp-modbus component folder: /home/codiak/lilygo-micropython/build_dir/esp32s3/esp-idf/components/freemodbus.
Including User C Module(s) from /home/codiak/lilygo-micropython/extmod/micropython.cmake
CMake Error at /home/codiak/lilygo-micropython/extmod/display/lcd/src/micropython.cmake:3 (include):
  include could not find requested file:

    /home/codiak/lilygo-micropython/build_dir/esp32s3/build-LILYGO_T-DisplayS3/../lcd_binding_micropython/lcd/micropython.cmake
Call Stack (most recent call first):
  /home/codiak/lilygo-micropython/extmod/micropython.cmake:19 (include)
  /home/codiak/lilygo-micropython/build_dir/esp32s3/micropython/py/usermod.cmake:42 (include)
  main/CMakeLists.txt:14 (include)_

closed 2024-05-18 by github-actions[bot]