Xinyuan-LilyGO/T-Display-S3 #131: T-Dsiyplay-S3 Touch  - 3D files housing

369Martin369, 2023-03-26:
fyi: 
I designed a case for beloved touch.... 2 prints only....works on FDM printer with 0,4mm nozzle
https://www.thingiverse.com/thing:5933982/files
https://a360.co/42BD9iV

bratoff, 2023-03-26:
How would you feel about sharing the source files in a git?  That would allow others to tweak as necessary for their printers.

369Martin369, 2023-03-27:
here we go, Fusion360 format
[T-Display-S3-Touch_V19.zip](https://github.com/Xinyuan-LilyGO/T-Display-S3/files/11075311/T-Display-S3-Touch_V19.zip)

369Martin369, 2023-03-28:
please follow project in Thingiverse

bratoff, 2023-03-28:
> please follow project in Thingiverse

Thingiverse contains only the STLs.  I was asking about the files from whatever CAD tool you used to create the STL.  That would facilitate making any necessary tweaks to get it to print on a different printer.

369Martin369, 2023-04-01:
> > please follow project in Thingiverse
> 
> Thingiverse contains only the STLs.  I was asking about the files from whatever CAD tool you used to create the STL.  That would facilitate making any necessary tweaks to get it to print on a different printer.
> 

Files were added. Have fun

lewisxhe, 2023-04-03:
@369Martin369 Thank you for your contribution. Can I add links or files to the warehouse's homepage?

369Martin369, 2023-04-03:
> @369Martin369 Thank you for your contribution. Can I add links or files to the warehouse's homepage?

Sure, however you want...also source files from zip file

lewisxhe, 2023-04-03:
Thank you, I have added and explained the source

closed 2023-04-03 by 369Martin369