Xinyuan-LilyGO/T-Display-S3 #135: Img2LCD Tool does not allow conversion of large images

PlastiBots, 2023-04-03:
Hi,
I tried using the Image2LCD [tool ](https://github.com/Xinyuan-LilyGO/T-Display-S3/blob/main/tools/Image2Lcd.zip) and it would not allow an image of 1950x240 to be converted. It forced it to 240x240.  Can this be looked into?  I am attempting to use a single wide image to handle multiple images for use in a Sprite from Matt's great work [here](https://github.com/h5n1xp/Arduino_Sprite).

lewisxhe, 2023-04-06:
Image2LCD is not developed by LilyGo, it's just a free tool on the Internet.

Is it possible to divide a large picture into multiple small pictures to ensure that it can be used correctly?

PlastiBots, 2023-04-06:
Ok thx.  To your suggestion, yes it's possible, but I am doing the reverse on purpose for using with sprites. e.g. My target is 19 items in 1 image of 7200px x 240px that can be "shifted" using Sprites.  All good, I'll reach out to the author.

closed 2023-04-06 by PlastiBots