Xinyuan-LilyGO/T-Display-S3 #136: LilyGO T-Display-S3 Touch and I2C

Archangel95, 2023-04-05:
Hi, I'm a bit confused about the use of the I2C with the Touch variant of this board. This is what I mean:

according to the pin schematics, the SDA and SCL are connected both on the JST pins (43 and 44) but at the same time, in order to work, the touch is connected with different SDA and SCL pins (17, 18).

So, my question is, which pin should I use in order to use both the touch and other devices connected to the JST connector (or pin 43,44), because right now I can't use both and Wire's library is giving me many errors ( for example: i2cRead returned Error -1).

Any ideas?
Thanks

lewisxhe, 2023-04-06:
Most sensor libraries use Wire by default, so you can change touch to Wire1 so that the sensor library calls Wire by default

You can also explicitly call Wire1 and initialize it
Wire1.begin(sda,scl)

When using the sensor library, pass Wire1 to the used library.

closed 2023-04-06 by lewisxhe

summerfind, 2024-07-18:
did you got solution? is the T-Display's I2C any difference from DEV Module?