Xinyuan-LilyGO/T-Display-S3 #137: What does brushing in a binary do?

teastainGit, 2023-04-05:
Some people feel that they have bricked their LilyGO T-Display S3 and Lewis He advice is to brush in a binary, perhaps a "BootLoader"?
I thought that the ESP32 S3 had a permanent indestructible BootLoader burned into ROM when the chip was made by Espressif?
What does brushing in a BootLoader do, exactly?
Thanks, Terry

lewisxhe, 2023-04-06:
esp32 属于两级引导模式，一级引导属于不可更改，所以esp32 永远都不可能刷成砖.

写入的bootloader属于二级引导程序，如果用户在写入固件过程中断开或者写入错误的文件，那么重新写入二级引导程序确保二级引导是正确的。

如果你关心过烧录过程，就可以发现每次上传草图的时候都会上传 bootloader.bin
[image]

closed 2023-04-06 by lewisxhe

teastainGit, 2023-04-06:
Thank you Lewis
-Terry