Xinyuan-LilyGO/T-Display-S3 #139: Error: "esp_lcd_i80_bus_config_t' was not declared in this scope"

bartjanisse, 2023-04-12:
Hi All,

After udating my Arduino IDE to version 2.0.4 ia am trying to recompile the Factory example code.

Now I get the following error message: "error: 'esp_lcd_i80_bus_config_t' was not declared in this scope"

Is there something i am missing here or has it to do with the new IDE.

I hope someone can help me out here.

Thanks a lot in advance,

Bart

bartjanisse, 2023-04-12:
Hi Terry,

Thank you for your quick response.

My board defs "ESP32 by Espressif, ver 2.0.7”. My libraries are installed at: "C:\Users\bartj\Documents\Arduino\libraries"

And yes my board (and compilation) was working fine before I updated the IDE. Saddly I don't know which I had before, it could be version 2.0.1 but I am not sure.

And yes I did an Arduino Library update after installing 2.0.4

Any thoughts that might help me?

Thanks again

teastainGit, 2023-04-12:
Bart, I did hear problems with ESP32 v2.0.7, so stick with 206, it works for me.
You must download the LilyGO T-Display S3 Zip file from their GitHub repository and copy TFT_eSPI to your library and overwrite the existing TFT_eSPI library. Do NOT use the Arduino generic library and DO NOT do a library update.
The recent Bodmer library is very good and useable on LilyGO T-Display S3 but you have to go deep into the file and comment out one board and un-comment board setupf # which is confusing, so for now just re-install the TFT_eSPI library!

Cheers !
Terry Stainton

***@***.***

bartjanisse, 2023-04-16:
Hi Terry,

Sorry for my late reply but I had some settings wrong in the Tool menu. Everything seems to be fine again.

Thank you so much for your input.

Kind regards,
Bart

369Martin369, 2023-05-20:
I use 2.0.9 without any problem...

closed 2024-05-16 by github-actions[bot]