Xinyuan-LilyGO/T-Display-S3 #140: Powering the board from 5V pin

dima-fedotovs, 2023-04-16:
Recently I was asking LilyGo team (via the chat on lilygo.cc) about powering the T-Display-S3 board from 5V pin. 
They answered, that it is designed to power the board from the 5v pin. So it is absolutely save to do so.
I just tested this just now and seems it doesn't work. 
What I did:
1. I've uploaded software via USB. 5V pin is not connected at this moment. Everything works fine.
2. I've unplug the USB and attach 5V via 5V pin and GND pin. There are no other connected pins and no USB cable attached.
Powering from 5V pin produces very quiet high frequency noise sound. Screen is black and seems nothing is working.
Removing 5V pin and connecting USB - everything still works fine.

This sound shows that definitely something is wrong. And probably some part of the board is killed already. 

So the question is: should the T-Display-S3 board work if to power it from the 5V pin or only option to deliver the power is USB connector?

bratoff, 2023-04-17:
Please refer to this closed issue:  https://github.com/Xinyuan-LilyGO/T-Display-S3/issues/79 which contains a link to another git containing much useful setup information for the T Display S3.

In particular, take a look at Solution A in his setup.txt document - that should solve your problem.

dima-fedotovs, 2023-04-17:
Thank you! This solved my issue!
Writing HIGH to output 15 does the trick. I've overseen this pin on the schematic. But now makes sense.

But the high-frequency noise is worrying me. It appears if I'm powering the board from a 5V pin and output 15 is LOW. So this happens every time board is switched on until fully initialized. A very short period of time, but it still doesn't sound good.

closed 2024-05-15 by github-actions[bot]