Xinyuan-LilyGO/T-Display-S3 #141: S3 on Windows 10 - device mgr always shows the S3 as 2 separate JTAG/serial debug units

rpbgit, 2023-04-20:
no matter what button sequence try, the S3 always comes up in dev mgr as two "USB JTAG/serial debug unit" devices.

how do i get it to appear as COMn port ??? 

eventually, want to use one JTAG debug port for debugging/single stepping/breakpoints, and the other as download. 

but at this point its rather useless to me. i have two S3's and they behave the same way.

how to correct/fix ?

rpbgit, 2023-04-20:
dev environment is Microsoft Visual Code with Platform I/O

lewisxhe, 2023-04-21:
Press and hold the BOOT button, do not release it, then press RST, release RST, and finally release BOOT

rpbgit, 2023-04-21:
Hi Lewis,

Thanks for the reply... 

 

Did this many times, no joy... 

 

I finally used Zadig to force the USBSER.sys driver to be installed on the first JTAG port, now I have a COM port back.... but even though I see a pile of stuff coming over the serial port, it seems like its simply spinning in circles.  And I cannot break out of it by any combination of button reset presses/sequences.

 

I have two objectives – 

1.	Get get back to being able to upload code to the S3.  I was able to do this ONCE and now it seems like its bricked... 
2.	Get the jtag debugger working from platform i/o on the second port, which is supposed to work.  My understanding is that the JTAG debugger must use libusb? 

 

Maybe I installed the USBSER driver on the wrong port ? 

 

When I connect a term emulator to the port it keeps spinning in this circle... like a WDT or something is expiring.   When I hold the reset button, the COM port disappears as one would expect, release it ant it gets enumerated again.

 

-bob

rpbgit, 2023-04-21:
[image]

this is the loop the S3 seems stuck in when connected to the COM port

lewisxhe, 2023-04-22:
I have not used the jata function of esp32s3, please refer to the official guide for specific usage.
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/jtag-debugging/index.html
It can be seen from the output log that the chip is abnormal. It may be that the program is abnormal and causes infinite restarts. You can flash the factory firmware to check whether the chip can work normally.
https://github.com/Xinyuan-LilyGO/T-Display-S3/tree/main/firmware

369Martin369, 2023-05-20:
I encountered this problem once....no problem. get you IDE settings right. Put the S3 into "waiting for download" position. And flash with correct IDE settings which will switch of USB behaviour.

closed 2024-05-15 by github-actions[bot]