Xinyuan-LilyGO/T-Display-S3 #143: Unable to power board with wall plug, but computer USB-C works!

jessedmatlock, 2023-04-23:
I've been programming and testing the board using a USB-C cable connected to my computer - typical setup.. nothing fancy (Apple iMac, USB port on the keyboard)

BUT, when I connect the SAME cable to any wall wart (power adapter), the board fails to load or it loads the initial image (setup())  and doesn't run the loop().

I've tried several 5v wall power adapters, and which prove good when powering other USB-C devices, with the same cord I'm testing the display with.. ..

Any thoughts?

jessedmatlock, 2023-04-24:
Further testing, I used the same USB-C cable and a large battery bank (used for charging cell phones on the go).. and the display works. 

So, in closing,.. no 5v power adapter, regardless of wattage capability, will power the display fully.

Anyone else having this issue? Any suggestions?

jessedmatlock, 2023-04-24:
> I want to help but It seems an impossible situation! I have several for 7 months(!) and they will power from anything, including iPhone white cube chargers, so, I wonder if there is some upload in your LilyGO T-Display S3 that is the problem?
> Have you uploaded a simple known good sketch that that used to run when you got it?
> What do you think has changed?
> -Terry

Thank you! Great suggestion.. I uploaded a known good sketch, set GPIO15 to HIGH, it powered on PC USB as well as Wall Plug just fine.. no issues. 
I will work with the programmer to address the code and see if we can find what is causing this..

teastainGit, 2023-04-24:
Good news!
I have a community resource on GitHub here:
Https://github.com/teastainGit/LilyGO-T-display-S3-setup-and-examples 
You may find something useful!

Cheers !
Terry

richp-tech, 2024-01-28:
May I ask a question? I can't get the Lilygo T Display to mount in my MacBook Air. Any ideas? I have installed both drivers mentioned in the GitHub. I can see the device in the USB tree. Thanks.

[image]

[image]

teastainGit, 2024-01-28:
You have a 'TTGO' (!) not a LilyGO T-Display S3.
So use my GitHub repository with that in mind!
What programming environment are you using?

richp-tech, 2024-01-29:
Hello @teastainGit 

It's very kind of you to reply. The box says T display. My issue is getting it to mount on my MacBook Air. Are you saying I'm using the wrong drivers because I don't know what the board is? 

I'll take another look, thank you.

[image]

lewisxhe, 2024-01-29:
> Hello @teastainGit 
> 
> It's very kind of you to reply. The box says T display. My issue is getting it to mount on my MacBook Air. I installed the drivers which are in GitHub pdf for this device.
> 
> But it won't mount.
> 
> Thank you
> 
> [image]
> 
> https://github.com/Xinyuan-LilyGO/T-Display-S3/assets/59874838/da7875c0-90be-46db-86b5-3d3062f29830
> 
> [image]

https://github.com/Xinyuan-LilyGO/TTGO-T-Display

richp-tech, 2024-01-29:
Hi @teastainGit 
Thanks for your reply. I see. Sorry about that. Lots to learn and I misunderstood. No combination of button holds and clicks makes a difference. This is an M1 Macbook Air. Thanks for your help.
Rich

teastainGit, 2024-01-29:
With a modern Mac, one does not usually go to the Settings USB page to check if your ESP32 is connecting.
Are you using Arduino IDE ?
You can watch live connection, loss of connection right here:
[image]

and 

[image]

richp-tech, 2024-01-29:
Hi, that's what I expected. I was surprised I had to install drivers (I installed them both). It won't mount. I will show up in the terminal (twice) and in the USB tree (once) plus twice in Arduino. I have mashed it up somewhere.
[image]
[image]
[image]

richp-tech, 2024-01-29:
[image]

teastainGit, 2024-01-29:
The 'Board Info' feature on Arduino IDE where it tries to "Guess  Your Board"  is useless for ESP32 and serves as an upsetting distraction, solely.
Have looked at the Arduino IDE  "Select Other Board and Port"? whilst connecting, as I showed?
Also M1 Macs are very fussing about USB versus Thunderbolt. MY LilyGO products only connect using TB cables in the TB ports on the back...my older non S3 boards only connect to the native USB ports on the back.

richp-tech, 2024-01-29:
Hi @teastainGit I cracked it. It was my understanding of GitHub and libraries which you helped me fathom. A clean install made all the difference. Thanks very much! It was very kind of you. A huge leap forward in  my grasp of it. 
[image]

teastainGit, 2024-01-29:
Great to hear!!!
Cheers, -Terry

richp-tech, 2024-01-30:
Thanks again @teastainGit 
I'm just an artist with an interest in tech. You have helped greatly

closed 2024-05-07 by github-actions[bot]