Xinyuan-LilyGO/T-Display-S3 #148: LilyGo T-Display-S3 Showing no Display and not able to update the firmware.

apaar24, 2023-05-02:
I have T-Display-S3 I got the product few months back and i have been trying to get the display on but only the back light is turning on, even the example code tft wont work. I thought it is firmware issue but when I m trying to update the firmware it is showing an error "[ESP8266Loader_spi[1]][espDownloader.py][line:764][ERROR]: ESP32S3 Chip stub error esp_stub_and_set_baud.". Please Help me solve this issue or if replacing this device is possible.

teastainGit, 2023-05-02:
The "firmware" is The BootLoader and it is indestructible and resides in ROM, burned by Espressif when the chip was founded.
This board is very finicky about using the 'as shipped' libraries and support files!
You cannot use the Arduino IDE libraries, only the LilyGO supplied one!
I had a lot of trouble when I started my first LilyGO T-Display S3 so I made a resource here:
https://github.com/teastainGit/LilyGO-T-display-S3-setup-and-examples
Please read this top-to-bottom and get the HotHead example running...and go from there!
Let me know your progress!
-Terry

apaar24, 2023-05-03:
> The "firmware" is The BootLoader and it is indestructible and resides in ROM, burned by Espressif when the chip was founded. This board is very finicky about using the 'as shipped' libraries and [...]

Sure! I will take a look at it and try to work with the other libraries and resources. Will let you know how it goes. 
Thank you !

apaar24, 2023-05-03:
> The "firmware" is The BootLoader and it is indestructible and resides in ROM, burned by Espressif when the chip was founded. This board is very finicky about using the 'as shipped' libraries and [...]

I tried everything from your git and followed all the instructions but still I m not getting any display on my LilyGo-S3. The black light is turning on but there is no display. maybe the display is faulty. I don't know what to do now. Please help me!

apaar24, 2023-05-04:
> OK, can you try Erase All Flash Before Sketch Upload: "Enabled". This takes a long minute but will clear out the memory. It is very easy to get no display on the LilyGO T-Display S3 ! What version of TFT_eSPI are you using? Go to your libraries and enter the TFT_eSPI folder. The version is in "library.properties". Are you receiving Serial.print messages? This is a vital troubleshooting aid. Let me know what you get on Serial. Are you using Arduino IDE v2.1.0? ESP32 board def v2.0.6? -Terry

Erase all flash before sketch was enabled. Using "version=2.5.0" TFT_eSPI Library. Yes is it showing me serial messages after I uploaded your ESPNOW Sketch. This is what it is showing me in the serial monitor -

"ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4207033a
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbe4
load:0x403cc700,len:0x2a38
entry 0x403c98d4
In setup!"

Yes I m using Arduino IDE v2.1.0. Also when I Go to check the board Info it is always showing me some other board and it is always showing a different board whenever I open Arduino.

[image]

[image]

teastainGit, 2023-05-04:
Ok, some progress!
Unfortunately the ESP_NOW example does not have any display graphics, see line 1 & 2 of the sketch.
Try the HotHead example, it has rich graphics and motion.
The Arduino v2 has an annoying feature (or bug that needs fixing)  where it displays some random board info:
You must always go to this screen to be sure.
[image]
I think you are close, but these new S3 versions are very hard to set up.
The TFT_eSPI supplied from LilyGO has a new file called User_Setup_Select.h where you can check that you have the correct options.
You need setup 206
But it should be correct if it is the raw LilyGO LilyGO T-Display S3 library install.
-Terry

apaar24, 2023-05-06:
I had a look in the User_Setup_Select.h file and everything seems alright, it is showing all the drivers. Also how can I setup 206 can you please help me with that.
-Apaar

apaar24, 2023-05-06:
My Bad...

teastainGit, 2023-05-06:
If you copied the LilyGO lib files to your Libraries, then 206 is selected. Open the User_Setup_Select.h file and 'find' 206. 
It must be 'uncommented', like this: 
#include <User_Setups/Setup206_LilyGo_T_Display_S3.h>
If not reload all libraries from LilyGO LilyGO T-Display S3 GitHub  and overwrite your Arduino libraries as a group I.e. 'replace'.
I hope you get it running! 
But remember...skipping any step in the setup may result in no video.
Once you get it running, it will be stable.
-Terry

apaar24, 2023-05-06:
Yes sir it is uncommented. in the 
[image]
Arduino lib as well as the t-display file.

closed 2023-05-12 by apaar24

summerfind, 2024-06-09:
> Let me know your progress!
Hi
I got the hothead run, thing is did you test SD card?

epiHATR, 2025-02-24:
I was unable to use TFT_eSPI with SD card on  T-Display S3