Xinyuan-LilyGO/T-Display-S3 #149: Trying to make as large of an OTA partition table as possible (But when I flash it, I can no longer update OTA)

jshggagnier, 2023-05-06:
I'm looking to squeeze the most out of this little chip, and yet whenever I flash an image with this partition table it seems to fail at OTA transfers from then on (I have to hold boot in order to get it to flash properly)... It also fails on the default16.csv partition table as well.

I also don't get any image from the display when flashed with a bad partition, but I think I can guess why that happens since it seems to be unable to find the right areas.

Error messages:

> A serial exception error occurred: Cannot configure port, something went wrong. Original message: PermissionError(13, 'The device does not recognize the command.', None, 22)
Note: This error originates from pySerial. It is likely not a problem with esptool, but with the hardware connection or drivers.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
*** [upload] Error 1

My Custom partition table:
[Testpart.csv](https://github.com/Xinyuan-LilyGO/T-Display-S3/files/11413369/Testpart.csv)

lewisxhe, 2023-05-22:
Change the board to esp32s3box

jshggagnier, 2023-05-22:
> If you are using the Arduino IDE, change the board to ESP32 S3 DEV MODULE, and then flash select 16M PSRAM selection OPI

what if I am using platform.io in VScode? I used it since I was under the impression that it wasn't possible to include your own partition schemes in Arduino IDE

lewisxhe, 2023-05-22:
OTA example has been added  https://github.com/Xinyuan-LilyGO/T-Display-S3/blob/main/platformio.ini

closed 2024-05-05 by github-actions[bot]