Xinyuan-LilyGO/T-Display-S3 #151: Black screen with TFT

Cary55, 2023-05-19:
Hi, I always get a blackscreen with all the TFT examples I try.

The backlight works but nothing is displayed on the screen.

Test with the included TFT library and with version 2.5.0.

Test with arduino 2.1.0 and 1.8.18

arduino_gfx_demo and factory example works.

I need help please, if anyone can also send me working TFT example compiled to .bin format for me to try and see if it works.

Thanks

teastainGit, 2023-05-19:
Since Day-One this has been a problem with people, myself included, 'simply' trying to get a demo test program of some sort uploaded.
I am using Arduino v2.1 and ESP32 board def 206, I heard some problems with the latest.
Even though the TFT_eSPI by Bodmer is no longer forked, I recommend following LilyGO's instruction and copying all the files in their 'lib' folder into your libraries folder and 'replacing' I/e. destructive overwriting. not merging these files. Resist the temptation to do an 'Arduino update' !!!
There is a file in the TFT_eSPI directory called "User_Setup_Select.h" Be sure that style 206 is uncommented and is the only uncommented setup in the file. It is halfway down the file. (Also be sure to check that one of the first selections "// #include <User_Setup.h>  // Default setup is root library folder" is also commented-out!!!
Since I had a lot of trouble getting mine going 8 months ago I made a user support repository here:
https://github.com/teastainGit/LilyGO-T-display-S3-setup-and-examples. 
The HotHead example is designed to get a sampling of TFT features like push image, sprites, etc and it has a Serial.print() message for troubleshooting!
It is possible to skip or modify my setup procedure details, but for now, please follow them!
Hope this helps
-Terry

Cary55, 2023-05-20:
> Since Day-One this has been a problem with people, myself included, 'simply' trying to get a demo test program of some sort uploaded. I am using Arduino v2.1 and ESP32 board def 206, I heard some [...]

Hey, thanks for the reply.

I'm quite familiar with the TFT_eSPI library, so I have already performed the following manipulations..

- Version 206: Uncommented.
- User_Setup.h: Commented-out.

- Tried with LilyGO library and Bodmer version 2.5.0.

- Tried on a clean install of Arduino 1.8.18 and 2.1.0 (Virtual Machine)

I selected the Arduino settings as indicated on the github (ESP32S3 Dev Module and more..)

Always the same problem. No display just the backlight.

On the other hand I get an image with the other examples so my LCD works correctly.

I also found your github while looking for similar cases, and unfortunately it didn't help me.

If you have a .bin file of a sketch using TFT_eSPI that works for you, it would be nice if you could send it to me so I can make sure it works for me as well and that's a problem. from library somewhere.

Thanks for the help.

369Martin369, 2023-05-20:
sound so me, that you did not patch the libraries.....you need to install the shipped libraries and do not update them, unless you understand which files you need to copy manually after refreshing

369Martin369, 2023-05-20:
[esp32.esp32.zip](https://github.com/Xinyuan-LilyGO/T-Display-S3/files/11522201/esp32.esp32.zip)
here bin of my actual multimeter project

Cary55, 2023-05-20:
> sound so me, that you did not patch the libraries.....you need to install the shipped libraries and do not update them, unless you understand which files you need to copy manually after refreshing

I tried with the libraries on the Lilygo github AND with the official TFT_eSPI library in version 2.5.0 (which contains the line for S3 T-Display in User_setup)

Both don't work.

However, my problem seems to be more complex.

I have 2 S3 T-Display cards.

One works fine with the TFT example
The other does not work with the same sketch.

Both work with the factory and adafruit_gfx examples.

Your example doesn't work for me (on both cards)

Thanks

teastainGit, 2023-05-20:
So, if one board works fine, could you export that binary?
I am curious how do you burn a binary into an ESP32, sincere question!
If one board does not work it may have some remainder code snippet that is corrupted. To test for this upload the TFT sketch with “erase all memory n flash” set. It adds a long scary minute to the upload, but it can work.
Please let me know your progress, as dozens of users have said that my HotHead example was helpful.

Cary55, 2023-05-20:
> So, if one board works fine, could you export that binary? I am curious how do you burn a binary into an ESP32, sincere question! If one board does not work it may have some remainder code snippet that is corrupted. To test for this upload the TFT sketch with “erase all memory in flash” set. It adds a long scary minute to the upload, but it can work. Also can you receive Serial.print() messages to indicate that code is executing? Please let me know your progress, as dozens of users have said that my HotHead example was helpful.

Your "HotHead" example works on one of my two cards, but not the other.

The serial displays "tick/tock".

I tried "Erase All Flash Before Sketch" it didn't change anything.

I tried to directly upload the .bin of your "HotHead" example, doesn't work.

I use "flash_download_tool_3.9.4" for ESP32 S3 to upload the .bin file, and it works with the .bin from the "factory" example.
Configuration: https://imgur.com/Xm6SHvd

Do you think a hardware problem is possible? Knowing that the board works fine with the example "factory" and "adafruit_gfx" but nothing works using TFT_eSPI. (And it's not a library problem because it works perfectly for the other board.)

Thanks

teastainGit, 2023-05-21:
Well, this is a head scratcher !
When you upload HotHead to the bad board do you see the tick/tock?
The big difference between the "factory.ino" sketch and the "tft.ino" sketch is that factory uses lvgl exclusively and tft uses TFT_eSPI. (which you may already know)
But I don't understand how one board works and not the other!!!
So...that's all I got!
Keep us posted, maybe someone else has an idea.
-Terry

teastainGit, 2023-05-21:
Also...would you like me to post a .bin of the tft sketch?
I just figured out how to do that.
-Terry

Cary55, 2023-05-21:
> Well, this is a head scratcher ! When you upload HotHead to the bad board do you see the tick/tock? The big difference between the "factory.ino" sketch and the "tft.ino" sketch is that factory uses [...]

> Also...would you like me to post a .bin of the tft sketch? I just figured out how to do that. -Terry

When I realized that TFT_eSPI works on the other board I immediately try to upload the .bin on the bad board, same problem.

Yes the bad board displays tick/tock, the sketch works perfectly but without display while it works perfectly on the board.

Yes I know Lvgl, I know adafruit and I know TFT_eSPI, unfortunately it's TFT_eSPI that I want to use.

The strangest thing is that the bad board works perfectly with adafruit and lvgl, so the display is not corrupted/broken.

I contacted the support of the Lilygo site, I'm waiting for an answer...

Thanks

teastainGit, 2023-05-21:
Thanks for following up!
Best of luck, cheers.

lewisxhe, 2023-05-22:
Please try this [firmware ](https://github.com/Xinyuan-LilyGO/T-Display-S3/tree/main/firmware)

Cary55, 2023-05-22:
> Please try this [firmware ](https://github.com/Xinyuan-LilyGO/T-Display-S3/tree/main/firmware)
Hi,

- Doesn't work on the faulty card. (Blackscreen, backlight on)
- Works on the other card.

I noticed that the faulty card was not recognized as "ESP-32-S3" on arduino, it has no name just the port "COMX"

Thanks anyway

lewisxhe, 2023-05-22:
There is no doubt about it, it can be determined that it is a malfunction.

Cary55, 2023-05-22:
> There is no doubt about it, it can be determined that it is a malfunction.
> 
> Open Serial Monitor, what can you see? Can you see it start up?

Yes serial monitor report what it should be reporting, even on a sketch with no display I can see the return in the Serial Monitor.

The board behaves normally except the board name which I don't see in Arduino and no display when using TFT_eSPI.

Thanks

lewisxhe, 2023-05-26:
It can only be said that it is not normal.

Cary55, 2023-05-26:
> It is not normal.

Yes, not normal and strange.. I really don't understand why it works if I use other method than TFT_eSPI.

I need to use TFT_eSPI, otherwise I certainly would have never discovered this problem.

I contacted Lilygo support.. I'm waiting..

Anyway, thank you for trying to help me.

jalmince, 2023-06-17:
Hey Cary !

I d be interested to know if you found out what the pb was.
( I have the same kind of issues with the TFT_eSPI ! )
kind regards,
Jean-Paul

robcloot, 2023-06-27:
I have 4 T-Displays that use to work fine but all 4 screens died. Programs upload and the serial monitor works, but I now have a black screen on all 4 modules. Did I fry all 4? If I did, how?

lewisxhe, 2023-06-27:
@robcloot You may have updated the default configuration of TFT-eSPI through ide, please check if it is the same as your configuration

https://github.com/Xinyuan-LilyGO/T-Display-S3/blob/main/lib/TFT_eSPI/User_Setup_Select.h

Michaelmilz, 2023-06-28:
Same problem here. My Program works fine. Sends Temperature over MQTT. Touch ist working. 
Until Today the Display work correctly but now only backlight. Any answers ?

lewisxhe, 2023-06-28:
@Michaelmilz  Is it not displayed after re-updating the sketch?

Michaelmilz, 2023-06-28:
@lewisxhe Yes. I am uploading the sketch several times. I try to upload older versions of the sketch. No success. 
Uploading the sketch to another board works correctly.

robcloot, 2023-06-28:
I fixed it! Yes I did update the TFT_eSPI library and messed things up. Use the T-Display-S3-main/lib/TFT-eSPI folder in GitHub, and do not update the library.

teastainGit, 2023-06-28:
Hi! You can use the up-to-date Bodmer TFT_eSPI (version=2.5.30) if you edit the "User_Setup_Select.h" to comment-out the line:

//#include <User_Setup.h>           // Default setup is root library folder

and un-comment:

#include <User_Setups/Setup206_LilyGo_T_Display_S3.h>     // For the LilyGo T-Display S3 based ESP32S3 with ST7789 170 x 320 TFT

which is about halfway down the huge file of setups.
This way you can safely keep current without fear.
-Terry

robcloot, 2023-06-29:
It works now! Use the TFT_eSPI folder in Github and do not update the library.

WO4ROB

teastainGit, 2023-06-29:
So, did you try updating the TFT_eSPI in Arduino or from bodmer’s github, and edit the selection file?This procedure now works well, for the last couple months!Cheers !Terry

bgilreath, 2023-08-09:
One key thing I left off: 

GFX Library for Arduino by Moon On Our Nation.

In the working stack, it was version 1.3.1.

Version 1.3.5 works with esp32 2.0.11 and lvgl 8.3.9.

Versions 1.3.6 and 1.3.7 either return a black screen or some major degradation.

bsbrum, 2023-11-23:
> Since Day-One this has been a problem with people, myself included, 'simply' trying to get a demo test program of some sort uploaded.
> I am using Arduino v2.1 and ESP32 board def 206, I heard some [...]

^^^^THIS HELPED^^^^ Thank you

closed 2024-05-05 by github-actions[bot]