Xinyuan-LilyGO/T-Display-S3 #155: TFT only working when connected via USB to Power Source or PC

Noengineer, 2023-06-01:
Hi guys,

I am having this issue where I can´t use the display while I am powering the device via battery connector.
First I thought it was my code but I tried Volos' Snake game and it also just lights up the display when connected via USB. (https://github.com/VolosR/SnakeT-DisplayS3).
My code makes the TTGO T-Display S3 act as a simple remote control for another esp32, so the display is not necessary for the code to function, but it would be nice to have as an indicator for when the device is in sleep mode and when it is active. (and it´s pretty).

Greetings

teastainGit, 2023-06-01:
Please see my LilyGO T-Display S3 setup and examples:
https://github.com/teastainGit/LilyGO-T-display-S3-setup-and-examples 
This sounds like a case:
C) Can run on USB power but not battery connector:
	In declarations add
	#define PIN_POWER_ON 15		// just before the setup and then in setup add:
	pinMode(PIN_POWER_ON, OUTPUT); 	//"triggers the LCD backlight, and enables battery power
	digitalWrite(PIN_POWER_ON, HIGH);)	//on startup"
	LilyGO variously refers to them as "LCD power on" OR "PIN_POWER_ON")
Hope this helps!
-Terry

Noengineer, 2023-06-01:
Works like a charm

Noengineer, 2023-06-01:
> Please see my LilyGO T-Display S3 setup and examples: https://github.com/teastainGit/LilyGO-T-display-S3-setup-and-examples This sounds like a case: C) Can run on USB power but not battery [...]

Resolved

closed 2023-06-01 by Noengineer