Xinyuan-LilyGO/T-Display-S3 #156: Touch display working on esp-idf?

krupis, 2023-06-02:
Hello. I know that this project is intended to use for Arduino platform, but I am developing on esp-idf.
I can get the display to work without any issues, but the touch is not working.

Please see the below git issue regarding the touch display:
https://github.com/espressif/esp-idf/issues/11517

I used component from the esp-idf component registry:
https://components.espressif.com/components/espressif/esp_lcd_touch_cst816s

```
#define EXAMPLE_I2C_NUM                 0   // I2C number
#define EXAMPLE_I2C_SCL                 17
#define EXAMPLE_I2C_SDA                 18

    esp_lcd_touch_handle_t tp = NULL;
    esp_lcd_panel_io_handle_t tp_io_handle = NULL;

    const i2c_config_t i2c_conf = {
        .mode = I2C_MODE_MASTER,
        .sda_io_num = EXAMPLE_I2C_SDA,
        .scl_io_num = EXAMPLE_I2C_SCL,
        .sda_pullup_en = GPIO_PULLUP_ENABLE,
        .scl_pullup_en = GPIO_PULLUP_ENABLE,
        .master.clk_speed = 400000,
    };
    ESP_LOGI(TAG,"Initializing I2C for display touch");
    /* Initialize I2C */
    ESP_ERROR_CHECK(i2c_param_config(EXAMPLE_I2C_NUM, &i2c_conf));
    ESP_ERROR_CHECK(i2c_driver_install(EXAMPLE_I2C_NUM, i2c_conf.mode, 0, 0, 0));

    
    esp_lcd_panel_io_i2c_config_t tp_io_config = ESP_LCD_TOUCH_IO_I2C_CST816S_CONFIG();

    ESP_LOGI(TAG,"esp_lcd_new_panel_io_i2c");
    ESP_ERROR_CHECK(esp_lcd_new_panel_io_i2c((esp_lcd_i2c_bus_handle_t)EXAMPLE_I2C_NUM, &tp_io_config, &tp_io_handle));

    esp_lcd_touch_config_t tp_cfg = {
        .x_max = 320,
        .y_max = 170,
        .rst_gpio_num = 21,
        .int_gpio_num = 16,
        .levels = {
            .reset = 0,
            .interrupt = 0,
        },
        .flags = {
            .swap_xy = 0,
            .mirror_x = 0,
            .mirror_y = 0,
        },
    };

    ESP_LOGI(TAG,"esp_lcd_touch_new_i2c_cst816s");
    esp_lcd_touch_new_i2c_cst816s(tp_io_handle, &tp_cfg, &tp);
```

The error message returned:
```
[1B][0;32mI (537) LVGL_SETUP: Initializing I2C for display touch[1B][0m
[1B][0;32mI (537) LVGL_SETUP: esp_lcd_new_panel_io_i2c[1B][0m
[1B][0;32mI (537) LVGL_SETUP: esp_lcd_touch_new_i2c_cst816s[1B][0m
[1B][0;32mI (537) gpio: GPIO[16]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 [1B][0m
[1B][0;32mI (547) gpio: GPIO[21]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:2 [1B][0m
[1B][0;31mE (577) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(135): i2c transaction failed[1B][0m
[1B][0;31mE (577) CST816S: read_id(174): I2C read failed[1B][0m
[1B][0;31mE (577) CST816S: esp_lcd_touch_new_i2c_cst816s(84): Read version failed[1B][0m
[1B][0;32mI (577) gpio: GPIO[16]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [1B][0m
[1B][0;32mI (587) gpio: GPIO[21]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [1B][0m
[1B][0;31mE (597) CST816S: Initialization failed![1B][0m
```

clive0106, 2023-06-02:
You are some steps ahead of me.
I am trying to make this display work with esp-idf, at this point, without touch.
Can you please point me to any example to get me started :)

krupis, 2023-06-02:
> You are some steps ahead of me. I am trying to make this display work with esp-idf, at this point, without touch. Can you please point me to any example to get me started :)

You can use my example project to get started:

https://github.com/krupis/T-Display-S3-esp-idf

Just call one function at a time to see how different widgets are displayed:
```
     display_meter();
    // display_image();
    // display_window();
```

The code should compile and run without any issues on T-Display-S3 modules from LilyGo on the esp-idf v5.0.1

clive0106, 2023-06-02:
Thank you so much...
It was more than I could have expected 
:+1:

clive0106, 2023-06-06:
@krupis After your help, I have managed to get mine responding to clicking the two physical buttons :)
I haven't tried the touch interface yet.
The LGVL stuff is tricky, being all message based.

krupis, 2023-06-06:
> @krupis After your help, I have managed to get mine responding to clicking the two physical buttons :) I haven't tried the touch interface yet. The LGVL stuff is tricky, being all message based.

Glad to hear. I am still hoping that someone managed to get the touch to work on esp-idf :) The arduino IDE examples from LilyGo seem to work fine with the touch

clive0106, 2023-06-08:
Or, is the stuff in menuconfig being ignored?

krupis, 2023-06-08:
@clive0106 
The example project in my repository that I have refered does not have anything to do with the CST816 touch.
It just has the simple LVGL implementation for the T-Display-S3.

As you have noticed, the esp-idf does not officially support CST816 touch driver (it is not in the menuconfig). You are supposed to download and import the driver via component registry (exactly like you did) and then you are on your own to get it going.

krupis, 2023-06-08:
> Was your project created using SquareLine studio?

Not this particular example.

But I have tried SquareLine studio and I was able to copy and paste the source files generated by SquareLine studio to my project and run the UI successfully.

krupis, 2023-06-14:
@clive0106 
Happy to announce that I managed to get Lilygo module touch working on the esp-idf with the help of esp-idf developers.

I have updated my repository for the example project:
https://github.com/krupis/T-Display-S3-esp-idf

Good luck and let me know if that works as expected. I still need to add some basic documentation for the repository.

skUlisesRostro, 2024-04-27:
Hello, I thank you very much for this information, I have been able to test the functionality of my Lilygo S3 touch, I only have one question, the manufacturer has told me that the cst816 chip recognizes up to 2 touches but I have not been able to test that function with any of the examples in the repository . Have any of you managed to recognize 2 points with this display?

closed 2024-06-10 by github-actions[bot]