Xinyuan-LilyGO/T-Display-S3 #158: Can't manage to use I2S  on the T-displayS3

jalmince, 2023-06-08:
I am trying to port code which runs fine on the T-embed, using its embeded mic and speaker.

Therefore I am trying to connect an  I2S mic and an I2S sound boards to the T-Display-S3.

pb: it looks like as soon as I initialize I2S, either on I2S_NUM_0 or NUM_1,  an other task displaying sprites that I have on core 0 ) freezes.

I suspect conflict with the display, but I am lost !

Could you please tell me if it possible to connect I2S devices on the T-Display, and if it is,  point me to some examples ?

with kind regards,
JPaul

teastainGit, 2023-06-08:
The T-Embed uses a SPI  LCD driver and the LilyGO T-Display S3 uses 8 bit parallel. Several of the T-Embed microphone pins are used in the LilyGO T-Display S3 LCD display, causing a conflict. (e.g. 47 & 48)
The T-Embed GitHub repository is here:
https://github.com/Xinyuan-LilyGO/T-Embed
Check it out for conflicts.
Also: The S3 supports I2S !!!

jalmince, 2023-06-08:
Thank you Terry, but:

> The T-Embed uses a SPI LCD driver and the LilyGO T-Display S3 uses 8 bit parallel  

-> I know

 Several of the T-Embed microphone pins are used in the LilyGO T-Display S3 LCD display, causing a conflict. (e.g. 47 & 48) 

-> I don t need to re-use the same pins   and want an I2S mic on the free pins 

Check it out for conflicts.  -> that was my question!!

Also: The S3 supports I2S !!! -> you have an example ?

Cheers, Jpaul

teastainGit, 2023-06-08:
I do not have an example. 
Leave this issue up and maybe someone else can help.

lewisxhe, 2023-06-19:
Can be used, how do you use it? Can you show me your code?

jalmince, 2023-06-19:
Hello Lewisxhe ! and thanks for your reply.

Actually I have mistitled my post, since it is about a conflict between I2S and the display using TFT_eSPI.

background 

I am porting onto the LilyGo T-Display-S3 code which runs fine on the LilyGo T-embed ( using its embeded mic and speaker, while displaying animated sprites)
Therefore I connected an I2S mic and an I2S sound boards to the T-Display-S3.
( code and connections are ok, since I can listen to a web radio, or read the mic samples.)

Here below code example  showing my issue:

the code "flashes" the display by alternating tft.fillScreen( TFT_BLACK) and tft.fillScreen( TFT_RED) in the loop.
of course it works absolutely fine.
But if I just try to install the I2S driver for the mic, ( without even reading the I2S), "flashing" is gone ...

I suspect a conflict, but ive struggled for days now, and can't find the solution.
code below if you have time to help ....

cheers,
Jean-Paul
```
#include "Arduino.h"
#include "pin_config.h"
#include <driver/i2s.h>

#include "TFT_eSPI.h" /* Please use the TFT library provided in the library. */
/* The product now has two screens, and the initialization code needs a small change in the new version. The LCD_MODULE_CMD_1 is used to define the
 * switch macro. */
#define LCD_MODULE_CMD_1

TFT_eSPI tft = TFT_eSPI();

#if defined(LCD_MODULE_CMD_1)
typedef struct {
    uint8_t cmd;
    uint8_t data[14];
    uint8_t len;
} lcd_cmd_t;

lcd_cmd_t lcd_st7789v[] = {
    {0x11, {0}, 0 | 0x80},
    {0x3A, {0X05}, 1},
    {0xB2, {0X0B, 0X0B, 0X00, 0X33, 0X33}, 5},
    {0xB7, {0X75}, 1},
    {0xBB, {0X28}, 1},
    {0xC0, {0X2C}, 1},
    {0xC2, {0X01}, 1},
    {0xC3, {0X1F}, 1},
    {0xC6, {0X13}, 1},
    {0xD0, {0XA7}, 1},
    {0xD0, {0XA4, 0XA1}, 2},
    {0xD6, {0XA1}, 1},
    {0xE0, {0XF0, 0X05, 0X0A, 0X06, 0X06, 0X03, 0X2B, 0X32, 0X43, 0X36, 0X11, 0X10, 0X2B, 0X32}, 14},
    {0xE1, {0XF0, 0X08, 0X0C, 0X0B, 0X09, 0X24, 0X2B, 0X22, 0X43, 0X38, 0X15, 0X16, 0X2F, 0X37}, 14},
};
#endif

// ---------------------------------------- I2S for microphone
// Connections to INMP441 
#define I2S_WS  1
#define I2S_SD  2
#define I2S_SCK 3 
#define I2S_CH_MIC I2S_NUM_1

void install_I2S_mic() {
  #define I2S_CH_MIC I2S_NUM_1
  #define SAMPLE_RATE 16000
    i2s_config_t i2s_config = {
      .mode = (i2s_mode_t)(I2S_MODE_MASTER | I2S_MODE_RX ),
      .sample_rate = SAMPLE_RATE,
      .bits_per_sample = I2S_BITS_PER_SAMPLE_16BIT,
      .channel_format = I2S_CHANNEL_FMT_ALL_LEFT,
      .communication_format = I2S_COMM_FORMAT_STAND_I2S,
      .intr_alloc_flags = ESP_INTR_FLAG_LEVEL1,
      .dma_buf_count = 8,
      .dma_buf_len = 64,
      .use_apll = false,
      .tx_desc_auto_clear = false,
      .fixed_mclk = 0,
      .mclk_multiple = I2S_MCLK_MULTIPLE_256,  
      .bits_per_chan = I2S_BITS_PER_CHAN_16BIT,
      .chan_mask = (i2s_channel_t)(I2S_TDM_ACTIVE_CH0 | I2S_TDM_ACTIVE_CH1),};
    i2s_pin_config_t i2s_mic_pins = {
      .bck_io_num = I2S_SCK,
      .ws_io_num = I2S_WS,
      .data_out_num = I2S_PIN_NO_CHANGE, 
      .data_in_num = I2S_SD};
    i2s_driver_install(I2S_CH_MIC, &i2s_config, 0, NULL);   
    i2s_set_pin(I2S_CH_MIC, &i2s_mic_pins);
    i2s_zero_dma_buffer(I2S_CH_MIC);
}

void setup() 
{
    pinMode(PIN_POWER_ON, OUTPUT);
    digitalWrite(PIN_POWER_ON, HIGH);
    Serial.begin(115200);
    delay(3000);
    Serial.printf("\nHello T-Display-S3\n");
    
    tft.begin();
    delay(500);
    #if defined(LCD_MODULE_CMD_1)
        for (uint8_t i = 0; i < (sizeof(lcd_st7789v) / sizeof(lcd_cmd_t)); i++) {
            tft.writecommand(lcd_st7789v[i].cmd);
            for (int j = 0; j < lcd_st7789v[i].len & 0x7f; j++) {
                tft.writedata(lcd_st7789v[i].data[j]);
            }

            if (lcd_st7789v[i].len & 0x80) {
                delay(120);
            }
        }
    #endif
    tft.setRotation(3);
    tft.setSwapBytes(true);

    ledcSetup(0, 2000, 8);
    ledcAttachPin(PIN_LCD_BL, 0);
    ledcWrite(0, 255);

    pinMode(I2S_SD, INPUT);
    pinMode(I2S_SCK, OUTPUT);
    pinMode(I2S_WS, OUTPUT);

    //install_I2S_mic() ;  //IF THIS ONE IS COMMENTED OUT, THE TFT FUNCTIONS GET IMPAIRED !!
    //                       ===============================================================
}

void loop(){
  tft.fillScreen( TFT_BLACK);
  delay( 500); 
  tft.fillScreen( TFT_BLUE);
  delay( 500);  
  Serial.print(".") ;
}
```

lewisxhe, 2023-06-21:
@jalmince  I have not arrived at the I2S barley wind, I am currently in the process of contacting you to provide a replacement, no problem, normal renewal of the screen.

jalmince, 2023-06-21:
Thanks for that ! Does it mean you could not reproduce the issue on one of your boards? And that my one is faulty ? That's weird since when using only the screen it looks fine.

Kind regards,
Jean-paul

yair-ehrenwald, 2023-10-02:
@jalmince 

Hi, just wondering, did you ever manage to resolve the issue? I ran into the same problem with the t-display s3.

yair-ehrenwald, 2023-10-03:
In case anybody else runs into this problem: 

While I couldn't find an actual reason for conflict with the I2S pins, changing them to use 43 44 and 18 works fine with the display.

closed 2024-05-04 by github-actions[bot]