Xinyuan-LilyGO/T-Display-S3 #160: Simple hello world example in RUST?

milewski, 2023-06-14:
Does anyone have a basic hello world example of working with the display written in RUST? 

I have no clue what is wrong with the example below, I suspect I have mapped the pins incorrectly, but I find no reference of what they should be anywhere online for this specific hardware:
 
```rust
fn main() -> anyhow::Result<()> {
    esp_idf_sys::link_patches();

    let peripherals = Peripherals::take().unwrap();

    let backlight = peripherals.pins.gpio38; // Pretty sure this one is correct
    let dc = peripherals.pins.gpio7;         // no idea what DC standards for... direct current?
    let cs = peripherals.pins.gpio6;         // Im in doubt if this is right.. or should it be GPIO10?    
    let rst = peripherals.pins.gpio5;        // No idea what RST stands for... mapped to LCD_RES (Resolution?)
    let spi = peripherals.spi2;              // I don't know if I should use SPI2 or 3 
    let sclk = peripherals.pins.gpio12;      // I idea what Something Clock? should map to.. so my guess was SPI_CLK
    let sdo = peripherals.pins.gpio11;       // Clue less... no mention of what SDO means anywhere on everywhere I look

    let mut backlight = PinDriver::output(backlight)?;
    backlight.set_high()?;                   // Geat! The backlight turns on when I run the code

    let di = SPIInterfaceNoCS::new(          // Found this example here: https://github.com/ivmarkov/rust-esp32-std-demo/blob/50ab64ed4bf51f02807864458e096c595b94cc1c/src/main.rs#L1045-L1071
        spi::SpiDeviceDriver::new_single(
            spi,
            sclk,
            sdo,
            Option::<gpio::AnyIOPin>::None,
            Some(cs),
            &spi::SpiDriverConfig::new(),
            &spi::SpiConfig::new()
        )?,
        gpio::PinDriver::output(dc)?,
    );

    let mut display = mipidsi::Builder::st7789(di)
        .init(&mut delay::Ets, Some(PinDriver::output(rst)?))
        .map_err(|e| anyhow::anyhow!("Display error: {:?}", e))?;

    display.clear(RgbColor::RED).unwrap();   // have tried various ways of drawing anything to screen but LCD always black

    loop {
      
    }
}
```

closed 2023-06-16 by milewski

StefanWerW, 2023-08-17:
Hi @milewski, I have followed your example and it also worked for me, but it is extremely slow, around 1fps. Any ideas?

milewski, 2023-08-18:
Hi @StefanWerW yeah, I had the same issue, I posted this same question on another repo https://github.com/ivmarkov/rust-esp32-std-demo/issues/152#issuecomment-1595817789 and got some clue of what could be the issue (bit-banging) and there was a suggestion to switch to DMA... 

I did not go further down this path yet, but I am going to be revisiting this soon, if you find a solution please share :)

psytraxx, 2023-11-30:
@milewski  -thanks so much for sharing your solution -  i followed your example and the screen is cut off -did you experience the same problem ?  
Text::with_baseline("Hello world", Point::zero(), text_style, Baseline::Top) basically prints "o world"

milewski, 2023-12-01:
It wasn't cut off on my on tests, it was just slowwww... maybe you got a different version?

dunef-io, 2024-01-06:
You need to start the code with 'cargo espflash flash --monitor --release', or add

```toml
[profile.dev]
opt-level = 1 #  a number in the range of 1-3
```
The optimization level makes a huge difference in the execution performance.

lmnzx, 2025-02-13:
@milewski can you please share the entire main.rs and Cargo.toml as I tried to follow your code but couldn't get it working. Thanks.

will-lynas, 2025-07-19:
> [@milewski](https://github.com/milewski) -thanks so much for sharing your solution - i followed your example and the screen is cut off -did you experience the same problem ? Text::with_baseline("Hello world", Point::zero(), text_style, Baseline::Top) basically prints "o world"

What worked for me was setting a display offset:

```rust
let mut display = mipidsi::Builder::new(models::ST7789, display_interface)
    .reset_pin(reset)
    .display_size(DISPLAY_WIDTH, DISPLAY_HEIGHT)
    .display_offset(36, 0) // Not sure where this number comes from, but this is what works in testing
    .invert_colors(ColorInversion::Inverted)
    .init(&mut FreeRtos)
    .map_err(|e| anyhow::anyhow!("Display init error: {:?}", e))?;
```

will-lynas, 2025-07-19:
> [@milewski](https://github.com/milewski) can you please share the entire main.rs and Cargo.toml as I tried to follow your code but couldn't get it working. Thanks.

https://github.com/will-lynas/esp-projects/tree/b46e9152b9e2694663e996b635ac5500f385b7c4