Xinyuan-LilyGO/T-Display-S3 #161: JST pin 43 not useable with Dallas DS18B20 sensor

teastainGit, 2023-06-15:
I am trying to connect a Dallas Semiconductor  DS18B20 temperature sensor to either the GPIO43 or the JST pin 43, which are probably the same circuit.
Using Arduino v2.1.0.
There is no response from the sensor at all, the oscilloscope shows 3.3v steady.
The sensor works on GPIO 1 (or  GPIO 18, e.g.).
The pin 43 can be programmed as an input or an output successfully, but not the serial serial I/O of the DS18B20.
Further another user says that the JST pin 43 works in ESPHome, but I cannot get it to work on Arduino. His application of the LilyGO T-Display S3 is to use DS18B20 on pin 43 of the JST for compact packaging, I suppose. He may have to resort to ESPHome, but would rather use Arduino! I cannot understand why it works in only one software.
My question therefore is:
"Is there a configuration setup for the LilyGO T-Display S3 that configures GPIO43 for I2C only? How can we free it up?"
Thanks, Terry

lewisxhe, 2023-06-16:
T-Display-S3 43, 44, complete direct connection core, no other original equipment, I used a different plate before, and it's the same problem, I didn't have it, I got it. [arduino-esp32]（https://github.com/espressif/arduino-esp32）

patofoto, 2023-06-27:
@lewisxhe Apologies but not sure I understand your response. Did you get it to work?.

closed 2024-04-29 by github-actions[bot]