Xinyuan-LilyGO/T-Display-S3 #162: A fatal error occurred: Unable to verify flash chip connection (No serial data received.).

d3mac123, 2023-06-16:
Board was working fine until it started to give me this error: 
```
Sketch uses 886997 bytes (45%) of program storage space. Maximum is 1966080 bytes.
Global variables use 52960 bytes (16%) of dynamic memory, leaving 274720 bytes for local variables. Maximum is 327680 bytes.
esptool.py v4.5.1
Serial port /dev/cu.usbmodem223401
Connecting...
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 34:85:18:71:7e:78
Uploading stub...
Running stub...
Stub running...

A fatal error occurred: Unable to verify flash chip connection (No serial data received.).
Failed uploading: uploading error: exit status 2
```

Other times the error is:
```
 fatal error occurred: No serial data received.
Failed uploading: uploading error: exit status 2
```

How can I reset it? I have trying different cables, sketches, settings, everything but no success.

lewisxhe, 2023-06-16:
https://github.com/Xinyuan-LilyGO/T-Display-S3#quick-start

d3mac123, 2023-06-16:
I have tried it so many times without success (hold the Boot button, connect the USB cable, press RESET, release Boot, release Reset). I have noticed that the board time changes from ESP32S3 Dev module to u-blox NORA-W10.

lewisxhe, 2023-06-16:
No, insert USB, press and hold BOOT, do not release the boot button, then press RST. Finally, release the boot

d3mac123, 2023-06-16:
I am probably doing something wrong:
- I open Arduino IDE with my sketch
- I select board ESP32S3 Dev Module
- No serial port is select yet because the board is not connected
- I connect the board via the USB cable
- I press and hold the BOOT button
- While still pressing the BOOT button, I press RST
- I release the BOOT button
- I release the RST - when should I do that? 
- Board shows in Arduino IDE as u-blox NORA-W10 and auto-captures a serial port (cu.usbmodem223401)
- I run the code and get the same error message again :(

lewisxhe, 2023-06-16:
1. connect the board via the USB - cable 
2. press and hold the BOOT button
While still pressing the BOOT button,  press RST
3.release the RST 
4.release the BOOT button

d3mac123, 2023-06-16:
You are the best! Thank you!

closed 2023-06-17 by lewisxhe

Fabryz, 2023-09-14:
> No, insert USB, press and hold BOOT, do not release the boot button, then press RST. Finally, release the boot

For you that arrived here via search engines:
are you stuck in this loop?

```
Connecting......................................

A fatal error occurred: Failed to connect to ESP32-S3: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
Failed uploading: uploading error: exit status 2
```

The quoted message is the solution! This finally worked for me, thanks

jonatmudd, 2024-02-06:
Thank you, thank you! The Boot - reset - release worked for me too!

electricmotel, 2024-04-20:
Awesome!  It worked for me too.  
Its strange that this problem just presented itself and has occurred with multiple esp32-s3 xiao units,  Even sending the basic 'blink' program had the issue.  Smells like a windows update problem.  The problem started after an update,  Hmmm.

electricmotel, 2024-04-20:
update.....

> Awesome! It worked for me too. Its strange that this problem just presented itself and has occurred with multiple esp32-s3 xiao units, Even sending the basic 'blink' program hxperd the issue. Smells like a windows update problem. The problem started after an update, Hmmm.

Update....After some more testing I discovered the error occurred with both esp32s3 and esp32c3 boards.  I checked for windows updates and there was one available.  The update included a BIOS update too, but that is probably just a Dell thing.  Either way....after installing the update the problem has completely disappeared.