Xinyuan-LilyGO/T-Display-S3 #164: What are the hardware serial pins on the T-Display and how should I reference them?

contractorwolf, 2023-06-18:
I am having a little issue connecting from this board to another one over serial.  I thought I could use pins 18 (rx) and 17 (tx) but I dont seem to be getting any response using Serial1 connected at 9600.

This is what I am doing (sudo code):

`
     Serial.begin(9600);
     Serial1.begin(9600);
     Sleep(3000);
     Serial1.println("AT");
     Sleep(3000);
     if(Serial1.avalable()>0) {
         Serial.println(Serial1.readString());
     }
`  

but I am not getting a response, should I switch to software serial and use other pins?  Anyone had any lucky with serial here?

lewisxhe, 2023-06-18:
Serial1.begin(115200, SERIAL_8N1, 18, 17);

contractorwolf, 2023-06-18:
thanks @lewisxhe that got it working!

contractorwolf, 2023-06-18:
SOLVED

closed 2023-06-18 by contractorwolf