Xinyuan-LilyGO/T-Display-S3 #166: Lilygo T-Display S3 does not appear as USB device in Windows 10

david65536, 2023-06-26:
I am wondering if it's known what the exact system requirements are for USB-serial on Windows 10 since the device doesn't seem to appear as a USB Peripheral in the Device Manager.  

Using a USB-A to USB-C cable the T-Display S3 gets powered on, and I am able to successfully enter DFU mode using the button sequence, but this still does not result in the device appearing as a USB Device or any activity from Windows.  It does not appear in the Device Manager.  

Using a USB-C to USB-C cable, the T-Display S3 produces a warning in Windows Explorer - "USB device not recognized - the last USB device you connected to this computer malfunctioned.  Windows does not recognize it".  In this case it also does not appear in the Device Manager. 

Expressif has some information about [establishing serial communication](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/get-started/establish-serial-connection.html) with the ESP32-S3, as well as [how to monitor this in Windows](https://docs.espressif.com/projects/esp-idf/en/v4.4/esp32s3/get-started/establish-serial-connection.html), but unfortunately that hasn't helped.

teastainGit, 2023-06-27:
Are you using Arduino IDE?
If so try LilyGO’s recommended settings.

Setting | Value
-- | --
USB CDC On Boot | Enabled
CPU Frequency | 240MHz (WiFi)
Core Debug Level | None
USB DFU On Boot | Enabled
Events Run On | Core 1
Flash Mode | QIO 80MHz
Flash Size | 16MB (128Mb)
JTAG Adapter | Integrated USB JTAG
Arduino Runs On | Core 1
USB Firmware MSC On Boot | Disabled
Partition Scheme | Huge APP (3MB No OTA/1MB SPIFFS)
PSRAM | OPI PSRAM
USB Mode | Hardware CDC and JTAG

Also on my M1 Mac some boards will not be recognized on USB, only Thunderbolt!
And others…vice versa.
So, I know you have a PC, but there is obviously a system fussiness!

david65536, 2023-06-27:
I am using the PlatformIO plugin for VSCode.  Should I expect to see a USB Serial Device on Windows (or a /dev/usb-serial-device on mac)?  It builds without issue but upload seems to be fail because it does not detect the port (and times out):
`Creating esp32s3 image...
Merged 2 ELF sections
Successfully created esp32s3 image.
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-builtin, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: COM3
Uploading .pio\build\lilygo-t-display-s3\firmware.bin
esptool.py v4.5.1
Serial port COM3
Connecting......................................

A fatal error occurred: Failed to connect to ESP32-S3: No serial data received.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
*** [upload] Error 2
=============================== [FAILED] Took 37.74 seconds ===============================
`
Using the USB-C to USB-C cable (thunderbolt 3 rated) this is the error message in Device Manager (USB-A to USB-C had no activity).  
[image]

These are my current settings in platformio.ini and you can see that ARDUINO_USB_CDC_ON_BOOT=1 (enabled):  
`[env:lilygo-t-display-s3]
platform = espressif32
board = lilygo-t-display-s3
framework = arduino
lib_deps = bodmer/TFT_eSPI@^2.5.30
monitor_speed = 460800
build_flags = 
	-D ARDUINO_USB_MODE=1
	-D ARDUINO_USB_CDC_ON_BOOT=1    
`

teastainGit, 2023-06-27:
I use Arduino IDE v2.1.0. You might try this just to get it running.
Before I compile and upload I check that it is connected, the Arduino V2 has a bug that shows the board type that it detects, not the ESP32S3 Dev Module that it actually needs, so you have to watch every time!
While on that screen if I cannot connect I do the "hold boot, press and release reset, then release boot". I see the port every time.
I have a user "Setup and Examples" GitHub  repository here:
https://github.com/teastainGit/LilyGO-T-display-S3-setup-and-examples 
Maybe read the setup text and get it running on Arduino IDE and then if PlatformIO has some feature that Arduino lacks, you could then branch off and try running the same test example in PlatformIO.
-Terry

david65536, 2023-06-27:
Thanks Terry, I was able to follow your instructions for the Arduino IDE but it still seems to have the same issue of the device/port not appearing.  It says there is an ESP32S3 connected to COM3 but I don't think that is actually true (rather it means I told it that's what port it would be on).
[image]

I think I will go ahead and install a Linux boot partition but will keep this issue open and happy to try anything else.  It will be interesting to see if the problem goes away once I flash it with something other than the factory image (via Mac OS or Linux).

teastainGit, 2023-06-27:
Well, thanks for following suggestions!
From this state I would pulldown that menu that says (USB symbol) ESP32S3 Dev Module. You should see this:
[image]
Do you see the port?

david65536, 2023-06-27:
Below is what I get with the device in DFU mode (consistent with what the Device Manager was showing earlier, and COM3 not being the thing I want).  I restarted the computer and no change.  You're definitely seeing a USB-Serial device that I'm not (at least with Windows 10). 
[image]

david65536, 2023-06-27:
I don't see anything other than COM3 regardless of whether or not the T-Display S3 is connected/disconnected  (or connected and in DFU)
[image]

teastainGit, 2023-06-27:
Ok, ummm...that is a system problem with your rig, sorry to say. I have never seen "the wrong port" selected here.
Go ahead with Linux!
It's 1 AM here, I'll check in the morning!

lewisxhe, 2023-06-27:
1.Connect the board via the USB cable
2.Press and hold the BOOT button , While still pressing the BOOT button, 3.press RST
4.Release the RST
5.Release the BOOT button
6.Upload sketch
@david65536 Try this steps

david65536, 2023-06-28:
I was able to narrow the problem down to a bad USB-A to USB-C cable.  With another cable the device immediately showed up as COM5 when placed in DFU mode.

The USB-C to USB-C (thunderbolt 3) cable continued to show up as an unrecognized device.

closed 2023-06-28 by david65536

teastainGit, 2023-06-28:
David, thanks for narrowing down!I might be experiencing cable issues, unknowingly!Cheers !Terry

Closed #166 as completed.

ovendoor, 2024-06-25:
> 1.Connect the board via the USB cable 2.Press and hold the BOOT button , While still pressing the BOOT button, 3.press RST 4.Release the RST 5.Release the BOOT button 6.Upload sketch @david65536 Try this steps Watch out for changes in Windows Device Manager

@lewisxhe 
The code uploads now. I just can't get any of the demos to actually run!